makerbase-mks/MKS-Robin-Nano-V3.X #4: WIFI Support for maril UI Mode ...

themacboy, 2021-02-19:
Good morning,

I never liked the MKS UI, and sincerelly prefer the Marlin color UI.

Can you add support for wifi on Marlin?

Thanks in advance !>

[image]

acguardia, 2021-03-08:
You only enable wifi in configuration_adv.h on cards that already come with the soldered wifi module on the card.
In the case of MKS Robin Nano V3, just enable the LVGL UI display in "configuration.h".
If you want to use it in another graphical mode and have MKS WIFi module, enable on cconfiguration.h, only:
"#define SERIAL_PORT_2" on port 1 and install ESP3D V3 for the MKS card.
https://github.com/luc-github/ESP3D/tree/3.0

themacboy, 2021-03-09:
> You only enable wifi in configuration_adv.h on cards that already come with the soldered wifi module on the card.
> In the case of MKS Robin Nano V3, just enable the LVGL UI display in [...]

Good morning, I think you have not understood my message.

I know that using the MKS ui the WiFi works, but I don't like this one at all and I don't use them.

For this reason in this thread I am asking you to program the wifi module to work with Marlin ui as well.

acguardia, 2021-03-09:
As I said, you can use ESP3D.
I'm using Color UI with WIFI Robin Nano AND ESP3D.
[image]

themacboy, 2021-03-10:
> 
> 
> As I said, you can use ESP3D.
> I'm using Color UI with WIFI Robin Nano AND ESP3D.
> [image]

ummt hen I miss undertood you.

Im not skilled on that, then how is suposed I must install ESP3D v3, inside the Mks Marlin fork? 

Thanks in advance

trashypopashy, 2021-03-18:
> 
> 
> Let me try show you with pics:
> [image]

How do You install ESP3d V3 on the MKS WiFi V1.1?

trashypopashy, 2021-03-20:
Colud You provide a .bin-file, to flash via MKS Robin Nano V3 by renaming it to MksWifi.bn?

acguardia, 2021-03-25:
> @acguardia Colud You provide a .bin-file, to flash via MKS Robin Nano V3 by renaming it to MksWifi.bin, please? I already made one for my MKS Robin Nano V2, to Use the older ESP3D V2.1, but don't have the time, to redo it for ESP3D V3...

[MksWifi.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/6205685/MksWifi.zip)

FabioMC, 2021-03-25:
If I load this firmware sent by @acguardia  to the wifi module, the MKS-UI will work normally or it just works with Marlin Color UI?

trashypopashy, 2021-03-26:
> [MksWifi.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/6205685/MksWifi.zip)

Thank You for Your support! I could flash ESP3D V3 with Your file. I can connect to the MKS-Wifi-Module now and can see the ESP3D-V3-Menu.

There is one problem: I can not connect to the printer. Already flashed Marlin another time to the Robin Nano V3 with  "#define SERIAL_PORT_2 1" enabled an I also configured the ESD3D-V3 for Marlin with 250000 baud-rate.

trashypopashy, 2021-04-14:
Could someone tell us (FabioMC an me) how it is possible to usw the MKS_WiFi-Module in Color_UI, please? Mine does not comunicate with marlin.

FabioMC, 2021-04-14:
@trashypopashy. Updating a little my experience on that. I moved on and upload the WIFI firmware sent by @acguardia. The WIFI communicates by the WEB Interface with the printer and you can perform printer homing, or monitor the temperatures and other stuffs but independent of the UI used, the printer do not communicate with CURA, S3D, Pronterface or any other software that was supposed to communicate with. At the end I upload the original WIFI firmware and I am using the MKS-UI with the hardware which is not my best option since there are lots of menu's and submenu's on Marlin UI that I loose. While @makerbase-mks do not present us a solution I am afraid we will not be able to use the Color-UI.

trashypopashy, 2021-04-14:
Thank You for Your fast response. I do not even get the communication between Marlin with Color_UI an @acguardias Wifi-ESP3D-V3 working. Did You configure something special in Marlin?

FabioMC, 2021-04-14:
No. Just set the 2nd COM to 1 as you and upload the WIFI.bin into the MKS-WIFI hardware. I did not touch the Wifi options on Config_ADV (all disable). After that I upload the index.html.gz in the module web interface and then it was working with the limitations I mentioned before.

trashypopashy, 2021-04-14:
Hm, interesting... I did also not change any other Wifi-options in config_adv...

themacboy, 2021-05-05:
> [MksWifi.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/6205685/MksWifi.zip)

Thx mate !

One more question, have you change any pin definition before compile? Thx in advance !

Souls-Reaver, 2021-06-08:
Se sabe ya si funciona marlin ui, con el WiFi de serie? Tal y como. Hace con lvgl ui? Por favor actualicen más opciones en lvgl ui o dispongan de WiFi en marlin ui, gracias

SergioLOS69, 2022-08-01:
> [MksWifi.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/6205685/MksWifi.zip)

I made the changes you suggested ("#define SERIAL_PORT_2" on port 1) and Chose the color UI
No modifications on Configuration_adv.h
I put on the SD card the new Robin_nano_v3.bin and the MksWifi.bin file. 
I had the color UI but no wifi.
During the firmware flash no "updating wifi model firmware" had been shown.
Am I doing something wrong in the process?

trashypopashy, 2022-08-01:
You can only update the WiFi-module, if there is a firmware flashed on the Robin-Nano-V3 with LVGL-UI. Just take an old MKS-firmware for this board and flash ist, then flash the MksWifi.bin, then flash your new firmware to the board with something else than LVGL-UI.

ilko-k, 2024-03-22:
I tried it as described **[here](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/wiki/Use-MKS_Robin_WIFI-on-Robin-Nano)** but with no success. Then, after a bit of thinking, I realized it will not be practicle to me. Because the module will act as a hotspot, it means my PC would have to use wifi as well to connect to it and therefore forget about internet. That is not what I want.

ilko-k, 2024-08-09:
@acguardia hey, I'm very curious on how you did configure your ESP's firmware. Whatever I do, I can't get the wifi module to communicate with the printer in marlin mode