makerbase-mks/MKS-Robin-Nano-V3.X #5: USB drive and SD at same time?

themacboy, 2021-02-20:
Good morning, 

it is possible to use both at same time, USB drive and SD micro card?

I have tryed but din't success, not sure if I made a mistake, but never can work with both.

Specially critical when I has activatet USB drive in firmware, but try to update (again) firmware from SD card, first time looks perfect, but when restart the board tries to update firmaware again, but it was deleted, and board hang up.

I will try to make a video for you.

makerbase-mks, 2021-02-22:
@themacboy ,Hello, now you can either use USB stick or TF card at the same time. And actually, we are making the effort to make them to work together.

rhapsodyv, 2021-02-22:
Working on that! Soon will be released.

themacboy, 2021-02-24:
> 
> 
> @themacboy ,Hello, now you can either use USB stick or TF card at the same time. And actually, we are making the effort to make them to work together.

umm negatgive, if i plug my USB drive, then I dont have access to SD card.

A curios failure, try to plug a USB drive (dont remove it) and tgry to update the firmware using a SD card. In my board, the frist time it try to update the firmware, and aftert restart it tries again and again to update the firmeare, but it can't end and board is a brick. Then the only possible solution is update firmware form the USB drive.

Has any one noticed that problem?

rhapsodyv, 2021-05-30:
Color UI now supports multi volume too. https://github.com/MarlinFirmware/Marlin/pull/22004

themacboy, 2021-05-31:
Good morning, 

thanks for all your work.

One more question, then Mks repository is not updated? You recomend that we better use the offical Marlin repository?