makerbase-mks/MKS-Robin-Nano-V3.X #6: SD card not working

jorgejglopez, 2021-02-21:
Hi, I'm having problem with the SD card reader.
I'm able to print trough the usb, but when I try to print with a SD card nothing happen.

makerbase-mks, 2021-02-22:
Hello, @jorgejglopez , you can either use USB Stick or TF card at the same time now. And we are making effort to make them to be available at the same time. 
Now you can choose the USB or TF card before compile: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware#mks-robin-nano-v3x-build-and-update-firmware. And we also released several compiled file on: https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/tree/main/firmware/Marlin-bugfix2.0.x-MKS-2.1.2

jorgejglopez, 2021-02-22:
Ok, Thank you. I have another problem when I try to print by wifi. I'm trining to print with cura (with de plugin). All it's ok, I'm able to control the printer but when I try to print, the lcd show that an error happen. I need the sd card enable for this?

MKS-Sean, 2021-02-24:
Yes, the SD card is used to store gcode files.

closed 2021-02-24 by jorgejglopez