makerbase-mks/MKS-Robin-Nano-V3.X #7: NEOPIXEL Setup

FabioMC, 2021-02-22:
I have tried different PINS not in use (PA13 and PC4) to setup the Neopixel led's of my printer but no success so far. What should be the best PIN option to setup those?

Aritzobixo, 2021-02-23:
It would be interesting to know how to connect a led strip. It is a pity that this good board does not have a neopixel connection but it is possible to connect it to other great pins.

Aritzobixo, 2021-02-23:
> He probado diferentes PIN que no están en uso (PA13 y PC4) para configurar los LED de Neopixel de mi impresora, pero hasta ahora no he tenido éxito. ¿Cuál debería ser la mejor opción de PIN para configurarlos?

have you modified the .pins file?

FabioMC, 2021-02-23:
Yes. Have modified but no success. The output pins are limited. I found that one option is to use the Power_Off pin but than I will loose the configuration to turn off the printer when the print is finish. I would like an alternative for it.

Aritzobixo, 2021-02-23:
It would be nice if you tell us something about it. Because otherwise we lose

Aritzobixo, 2021-02-23:
@MKS-Sean any solution to the problem?

MKS-Sean, 2021-02-24:
@FabioMC @Aritzobixo Hello, PA8 (BL_TOUCH signal pin) or TX and RX pins of UART3 can achieve this function.

Aritzobixo, 2021-02-24:
[image]
and what pin is S, G, 5v ???

MKS-Sean, 2021-02-24:
S is PA8 (signal pin), you can see the pin number on the back of the motherboard

Aritzobixo, 2021-02-24:
> @FabioMC @Aritzobixo Hola, PA8 (pin de señal BL_TOUCH) o los pines TX y RX de UART3 pueden lograr esta función.

but that's for bltouch and i use bltouch too

MKS-Sean, 2021-02-24:
In this case you can use uart3's tx or rx pin

MKS-Sean, 2021-02-24:
[image]

Aritzobixo, 2021-02-24:
but that's for bltouch and i use bltouch too

Aritzobixo, 2021-02-24:
The image can not be seen

Aritzobixo, 2021-02-24:
If I use E1 driver, the connections are S = PA3, G = GND, and 5v? @MKS-Sean @makerbase-mks

FabioMC, 2021-02-24:
I would like to try TX RX UART 3 since looks more simple. What are the pin codes for those? PB10 and PB11? Is that correct?

FabioMC, 2021-02-24:
Ok. Tried UART 3. I used PB10 since its near the GND so an easier way to fit the connector. Everything is working properly. Thanks for your kind feedback.

Aritzobixo, 2021-02-24:
> Está bien. Probé UART 3. Usé PB10 ya que está cerca del GND, por lo que es una forma más fácil de colocar el conector. Todo funciona correctamente. Gracias por tus amables comentarios.

could you put a photo of the connections?  Thanks a lot

FabioMC, 2021-02-24:
[image]
Here it goes

Aritzobixo, 2021-02-24:
> [image]
> Aquí va

there you lose touch screen, right?

FabioMC, 2021-02-24:
Not really. I am using MKS-TS35 which uses only EXP1 and EXP2. I may not be able to use the new MKS H43 though. So far everything is working fine.

Aritzobixo, 2021-02-24:
> Realmente no. Estoy usando MKS-TS35 que usa solo EXP1 y EXP2. Sin embargo, es posible que no pueda usar el nuevo MKS H43. Hasta ahora todo está funcionando bien.

Could you put the piece of code that you have added, please?

FabioMC, 2021-02-24:
It is a very simple configuration. Enabled NEOPIXEL_LED and put the PIN number in front of NEOPIXEL_PIN. You don't need to change any other file

// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  **#define NEOPIXEL_PIN    PB10    // LED driving pin**
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 32       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS 127  // Initial brightness (0-255)
    #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif

Aritzobixo, 2021-02-24:
@FabioMC Hi Fabio, would you know how to do it with the uart pins? I do not understand where it says to connect the cables

FabioMC, 2021-02-24:
His suggestion is to use the PIN PD8 into the E1 step motor jumpers. Please note that 5V and GND need to be connect to other place like the MT-DET2 or any other connector that you are not using.

[image]

mitofun, 2021-04-06:
thank you folks everything works! but MKS should do the instruction for this!))))  PB10 it is TX3 pin under the processor!

rick090980, 2022-01-19:
Are you able to control the leds from the LCD with them set up this way, or only through g-code?

Freddyb888, 2022-06-28:
How many leds is maximum you can connect?

mitofun, 2022-06-28:
> How many leds is maximum you can connect?

hi) directly i think no more 8- if you ude the dcdc step down to 5 volts - when it is up to you)

heatchclif, 2022-06-30:
what version of marlin did you use????

Freddyb888, 2022-07-06:
> > How many leds is maximum you can connect?
> 
> hi) directly i think no more 8- if you ude the dcdc step down to 5 volts - when it is up to you)

I bought the DCdc step down how do I exa tly connect this and the 60 led ring? I dont want to break something. Thx

kabodos, 2022-07-30:
which ui are u using, marlin  or TFT_LVGL_UI?

Freddyb888, 2022-08-02:
> which ui are u using, marlin or TFT_LVGL_UI?

Tft