makerbase-mks/MKS-Robin-Nano-V3.X #8: PWM TTL For laser

Ainmimdal, 2021-02-23:
Hi, I need to supply 5v pwm signal to control my laser. Supposedly this board is said to have PWM TTL support compared to the robin nano V2, but I checked my board and the only empty pins I see are the three next to the fan LEDs. The silkscreen on the board says PA0 PE5 and PB0 which is the same for the heaters. I don't see any spindle/laser definitions in the example configs either.

Ainmimdal, 2021-02-23:
I tried using the PB0 and the PWM works fine for now. But can anyone confirm the voltage on this pin?

Ainmimdal, 2021-02-24:
After reading issue #7, Using TX3 (PB10) pin works. I still don't understand why the advertised PWM pins are the same pin for heaters.

MKS-Sean, 2021-02-27:
[image]
Hello, what I marked with a red box is the TTL pin that drives the MOS tube, which can be configured as a laser control pin

Ainmimdal, 2021-02-28:
Yeah, I tried those pins but the laser only goes up to 60% with those pins. Any components connected to the heater port will turn on if I use those pins. My solution is using the UART pins which outputs pwm correctly.

moibatista, 2021-05-24:
I have the MKS Robin Nano v3, with original configuration. I don't change anything.
How I can use de TTL pin? How I can active this pins? I need to know the Gcode to control this pin.
With M3 say me incorrect command.

Ainmimdal, 2021-06-05:
I use TX3 pin (next to the esp) and in marlin, you have to find the file pins_MKS_ROBIN_NANO_V3.h, find #define SPINDLE_LASER_PWM_PIN and change to #define SPINDLE_LASER_PWM_PIN PB10 . You can search more on the internet about changing the laser settings in configuration_adv.h

CLP58, 2022-04-12:
> I use TX3 pin (next to the esp) and in marlin, you have to find the file pins_MKS_ROBIN_NANO_V3.h, find #define SPINDLE_LASER_PWM_PIN and change to #define SPINDLE_LASER_PWM_PIN PB10 . You can search more on the internet about changing the laser settings in configuration_adv.h

I tried using the TTL PWM pin as they say, but found the same thing as you, only 60% power. How did you fix it please? I used the UART the same way you did and only had 60% power.

danorder, 2024-02-10:
did you change the jumper to 5v.