makerbase-mks/MKS-Robin-Nano-V3.X #10: I can't build last Marlin from MKS gith to Robin Nano V2/V3

acguardia, 2021-03-03:
I can't buid last marlin version to this board
I'm using #define TFT_LVGL_UI and all itens to Filament Runout Sensors:
#define FILAMENT_RUNOUT_SENSOR
#define FILAMENT_RUNOUT_SCRIPT "M600"
#define NOZZLE_PARK_FEATURE
#define ADVANCED_PAUSE_FEATURE
But, when I try to build return this errors
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\lcd\extui\lib\mks_ui\draw_error_message.cpp.o
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp: In function 'void btn_ok_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:167:7: error: 'pause_menu_response' was not declared in this scope; did you mean 'pause_menu_disp'?
167 | pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
| ^~~~~~~~~~~~~~~~~~~
| pause_menu_disp
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:167:29: error: 'PAUSE_RESPONSE_EXTRUDE_MORE' was not declared in this scope
167 | pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/boards.h:24,
from Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/MarlinConfigPre.h:37,
from Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:27:
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp: In function 'void btn_cancel_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:226:35: error: 'pause_menu_response' was not declared in this scope; did you mean 'pause_menu_disp'?
226 | TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
| ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
443 | #define THIRD(a,b,c,...) c
| ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
192 | #define __TERN(T,V...) ___TERN(_CAT(_NO,T),V) // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
| ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
191 | #define _TERN(E,V...) __TERN(_CAT(T_,E),V) // Prepend 'T_' to get 'T_0' or 'T_1'
| ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
190 | #define TERN_(O,A) _TERN(_ENA_1(O),,A) // OPTION converted to A or '<nul>'
| ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:226:5: note: in expansion of macro 'TERN_'
226 | TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
| ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:226:57: error: 'PAUSE_RESPONSE_RESUME_PRINT' was not declared in this scope
226 | TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
443 | #define THIRD(a,b,c,...) c
| ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
192 | #define __TERN(T,V...) ___TERN(_CAT(_NO,T),V) // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
| ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
191 | #define _TERN(E,V...) __TERN(_CAT(T_,E),V) // Prepend 'T_' to get 'T_0' or 'T_1'
| ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
190 | #define TERN_(O,A) _TERN(_ENA_1(O),,A) // OPTION converted to A or '<nul>'
| ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:226:5: note: in expansion of macro 'TERN_'
226 | TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
| ^~~~~
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\lcd\extui\lib\mks_ui\draw_dialog.cpp.o] Error 1
======================================================== [FAILED] Took 120.66 seconds ========================================================
Environment Status Duration
------------------------------------- -------- ------------
mks_robin_nano_v3_usb_flash_drive_msc FAILED 00:02:00.659

mks-viva, 2021-03-05:
Hi! You can refer to Wiki: https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/wiki/MKS_FES

xavierorrit, 2021-03-08:
It also doesn't let me compile the latest version of Marlin. He just leaves me with MKS.

closed 2021-03-08 by acguardia