makerbase-mks/MKS-Robin-Nano-V3.X #11: assets folder don't update

acguardia, 2021-03-03:
I downloaded the latest Marlin for MKS-Robin-Nano V2 / V3 and made all the changes to update the Robin Nano V3 card.
PlatformIO also compiled and generated "active" folder and .bin file, ok.
I pasted the files on the SD card and inserted it into the card.
When the update was made, only the .bin file was changed to .CUR, the "assets" folder remains "assets", but the correct thing would have been to change to "_assets".

------
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = mks_robin_nano_v3_usb_flash_drive_msc
include_dir  = Marlin
------

[image]

[Configuration.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/6077245/Configuration.zip)

ETE-Design, 2021-03-03:
It only use the assets folder if you have set the TFT to use MKS LVGL UI, else it dosen't use it...

Chlorel, 2021-03-05:
Hi, got the same problem when #define USB_FLASH_DRIVE_SUPPORT is uncommented ... 
But works fine when commented

mks-viva, 2021-03-08:
@acguardia Hi! Can you provide your firmware and assets to me?

acguardia, 2021-03-08:
> @acguardia Hi! Can you provide your firmware and assets to me?
Thanks for your answer.
Here are the files.
[assets.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/6104641/assets.zip)

theruleslawyer, 2021-03-13:
Are you using USB power? I was going nuts trying to figure out why my flashes kept failing. It looks like the SOC crashes because it can't draw enough power on USB. As soon as I hooked it up to my 24v power supply it was okay. I thought it'd be easier to bench test it prior to using big power.

Chlorel, 2021-03-13:
I used USB power at start, but it is the same error with external 12V power ... The asset only upload when USB_FLASH_DRIVE_SUPPORT  is commented

you find my setting & code here : https://github.com/Chlorel/Mks-Robin-Nano-Marlin2.0-Firmware/tree/Robin-Nano-V3

mks-viva, 2021-08-23:
The now, you can try to update it to the latest version firmware.