makerbase-mks/MKS-Robin-Nano-V3.X #14: Mismatched fan pin on documentation

ProjectPatatoe, 2021-03-20:
all revisions (001-003) has mismatched pin outs for Fan1 and Fan2. which is correct? It seems like the pin out got swapped between the FAN1 tag and the actual circuit/jack intended.

Images below are from 002 but the same or similar mismatch is on 001 and 003

* 002 BOTTOM.pdf
[image]
* 002 PIN.pdf
[image]
* 002 SCH.pdf
[image]

[image]
* 002 SIZE.pdf
[image]
* 002 TOP.pdf
[image]

mks-viva, 2021-03-21:
PC14 use for FAN1
PB1 use for FAN2
We fix it

mks-viva, 2021-03-22:
In the latest version 003 PCB, we have modified it 
[image]

ProjectPatatoe, 2021-03-22:
I ask for a clarification/fix because for 001 and 002 is because people who have an older revision board can't be sure what exactly they are seeing.

In addition, 003 still has inconsistencies. Specifically on the SCH sheet, they go to different connectors
* 003 BOTTOM.pdf
[image]
* 003 PIN.pdf
[image]
* 003 SCH.pdf
[image]
* 003 SIZE.pdf
[image]
* 003 TOP.pdf
[image]

themacboy, 2021-03-22:
Why all this revisions? there are manofacturing or design problems in revisiopn 1 and 2? Why must request a board update (change)?

mks-viva, 2021-03-23:
001 and 002 is design problems, so, we will fix it on 003 version

ProjectPatatoe, 2021-03-23:
All (001-003) versions have documentation that is not reflecting what people have, making diagnosing or tracking errors difficult. For instance, these are the full list of differences I can see from manually comparing all of the sheets. Which really this should have been just listed somewhere!

Revision1vs2

 - fixed silkscreen for "E0_MOTOR" (Bottom sheet)
 - missing TH1 silkscreen (Pin sheet)
 - add a diode array / resistor bottom right by J55 (Pin sheet)
 - reworded/rearrange heater page (Sch sheet)
 - added diode protection on BLTouch (Sch sheet)
 - added 100ohm on BLTouch servo pin (Sch sheet)
 - a few more measurements (Size sheet)

Revision2vs3

 - fan pin switch (Bottom sheet)
 - <del>clarified resist/silk/copper layers (Bottom sheet)</del> recent changed undid this ):

mks-viva, 2021-03-26:
001 and 002 is has silkscreen error and the actual PCB also has the same silk screen error, so this cannot be modified 
003 has fix all the silkscreen error and i has update sch

ProjectPatatoe, 2021-03-26:
Ok so those pin notations are actual silk screens and not just notes? its hard to tell because everything is the same color. Perhaps a note should be put next to it as someone diagnosing their board wouldn't know that its an error

mks-viva, 2021-03-27:
just like what you said, those pin notations are actual silk screens and not just notes. So, 003 version hardware has fix it.

ProjectPatatoe, 2021-03-29:
OK since its now relatively documented both here as well in the updated documents and readme, I will close the issue.

closed 2021-03-29 by ProjectPatatoe