makerbase-mks/MKS-Robin-Nano-V3.X #18: Virtual USB feature

Ainmimdal, 2021-04-02:
Somehow I can't get my computer to detect any storage devices when connected to the board via USB. I already changed the default environment in platform io to mks_robin_nano_v3_msc. I compiled my firmware using Auto Build Marlin but it only has two environments: mks_robin_nano_v3 and mks_robin_nano_v3_usb. I tried building mks_robin_nano_v3_usb_flash_drive_msc directly from platformio but the board keeps resetting when using USB. There are also no storage devices detected.

closed 2021-04-05 by Ainmimdal

spammchik, 2021-04-07:
> Somehow I can't get my computer to detect any storage devices when connected to the board via USB. 

Hello. How could you fix this?

123Li220, 2021-05-20:
> Somehow I can't get my computer to detect any storage devices when connected to the board via USB. 
Hi, I do have the same issue. Could you post here how you solved it?

Ainmimdal, 2021-09-02:
Sorry everyone I had a long break from 3d printing but if anyone still have this issue, I just cloned the freshest copy of the firmware and build using platformio using the msc environment. The usb data transfer works but very unreliable (loses connection and hangs my file explorer) that I never bothered to even try using anymore