makerbase-mks/MKS-Robin-Nano-V3.X #19: Endstop not work

alfaxx83, 2021-04-04:
Endstop not work

mks-viva, 2021-04-12:
Please add more information about "Endstop not work"

Jugin88, 2021-04-14:
hi, I'm new to the subject, my endstops don't work either, no matter what the pin assignment is, don't want to stop.

mks-viva, 2021-04-15:
@Jugin88 
1. Check endstop hardware is OK?
2. You need set endstop false or ture in configuration.h file

alfaxx83, 2021-04-15:
True or false change does not work. I endstop with LEDs. I have a printer Flying Bear Ghost 5

mks-viva, 2021-04-16:
Usually, endstop's S G connect to board's S G, you need change endstop logic in configuration.h file when homing have not stop.

lowtolerance, 2021-05-02:
If you follow the configuration check in the documentation, you should be able to resolve this problem on your own. You probably just  need to enable a pull-up or invert the logic, but klipper provides the tools for testing endstop functionality without actually attempting to home the axis.

closed 2021-08-03 by mks-viva

nfalco79, 2023-12-09:
I have the same issue and seems no configuration will resolve the issue. I have Flybear Ghost5, I had buy a Robin Nano 3.1. All endstops are ignored and the head try go over