makerbase-mks/MKS-Robin-Nano-V3.X #21: Not able to compile if using "M600" and "TFT_LVGL_UI"

t0mx, 2021-04-11:
Hello,

as I would like to be able to change filament I configured "M600" and was not able to use TFT_LVGL_UI as UI (TFT_COLOR_UI works):

`Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\lcd\extui\lib\mks_ui\draw_error_message.cpp.o
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp: In function 'void btn_ok_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:167:7: error: 'pause_menu_response' was not declared in this scope; did you mean 'pause_menu_disp'?
  167 |       pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
      |       ^~~~~~~~~~~~~~~~~~~
      |       pause_menu_disp
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:167:29: error: 'PAUSE_RESPONSE_EXTRUDE_MORE' was not declared in this scope
  167 |       pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/boards.h:24,
                 from Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:27:
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp: In function 'void btn_cancel_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:226:35: error: 'pause_menu_response' was not declared in this scope; did you mean 'pause_menu_disp'?
  226 |     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
      |                                   ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
  443 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
  192 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
  191 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
  190 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
      |                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:226:5: note: in expansion of macro 'TERN_'
  226 |     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
      |     ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:226:57: error: 'PAUSE_RESPONSE_RESUME_PRINT' was not declared in this scope
  226 |     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
  443 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
  192 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
  191 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
  190 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
      |                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:226:5: note: in expansion of macro 'TERN_'
  226 |     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
      |     ^~~~~
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\lcd\extui\lib\mks_ui\draw_dialog.cpp.o] Error 1`

Any idea?

ColdBeer72, 2021-04-13:
It's a classic... LVGL can't be compiled with M600. You must use substitute M600 by M25. I can't understand the reason to don't doing a bit of work here by MKS.

t0mx, 2021-04-14:
Thanks, good to know... Is there anywhere more information about this issue? How to substitute M600 by M25? Must ist also be known to the slicer or is it transparent for the slicer?

ColdBeer72, 2021-04-15:
You need to change this in **configuration.h**:
`#define FILAMENT_RUNOUT_SCRIPT "M600"`
by this:
`#define FILAMENT_RUNOUT_SCRIPT "M25"`

It's not transparent for the slicer. I use **CURA**, so you must go to **extensions // postprocessing // modify gcode**. Right there, there is an option called "Filament change", but you can't use it now, because it uses M600. But you could use **"Pause at Height"**, where you could select which type of pause to use.

t0mx, 2021-04-17:
Thanks, maybe I'll give them a chance but currently I'm doing well using Marlin Color UI.

Hopefully MKS can fix this problem soon!

artdezigner, 2021-11-30:
Друзья, включение M600 возможно для платы Robin Nano v2 +ts35? Скажите пожалуйста как сделать!

n9jcv, 2022-03-17:
@MKS-Sean @mks-viva 
Is anyone working to enable M600 with the Robin Nano V2 ?

rouhit110, 2022-06-22:
I have the same problem

t0mx, 2022-06-22:
I think, it won't be changed anymore.

n9jcv, 2022-06-22:
Well if they do not fix, vote with your wallets.  Do not buy MKS in the future

t0mx, 2022-06-23:
That's also my opinion. They make good electronics, but are not able to complete their work to integrate their UI in Marlin.

But they could at least state: we are not going to fix that, use either the workaround using M25 or don't use TFT_LVGL_UI at all.

barkh, 2022-10-29:
eneble EMERGENCY_PARSER and HOST_PROMPT_SUPPORT - probably marlin fault, there is OR in comment, but AND in  Conditionals_post.h, line 3433 (marlin 2.1.1), same line is probably responsible for not recognising TFT.

danielscatigno-ncpc, 2023-03-30:
I Have enabled NOZZLE_PARK_FEATURE, EMERGENCY_PARSER, HOST_ACTION_COMMANDS and HOST_PROMPT_SUPPORT
in Configuration.h and Configuration_adv.h
and at least it is compiling
I'll be checking if it is working

ysnuysl, 2023-12-27:
I'm stuck at this part too and I'm getting the same error. I updated the part from "M600" to "M25", but it still didn't work, I got the same error. What am I missing?

Mr-Tri-Luc, 2024-06-01:
You miss a little step. You have to use "TFT_COLOR_UI" with Filament runout detection

schnurly, 2024-10-11:
i had the same problem ,solved it by removing the condition for #include "../../../feature/pause.h"
in mks_ui/draw_dialog.cpp

mrbrain-true, 2025-06-01:
I made the following correction in the file Marlin\src\feature\inc\Conditionals_posts.h:

/**
 * Advanced Pause - Filament Change
 */
#if ENABLED(ADVANCED_PAUSE_FEATURE)
#if ANY(**_HAS_TFT_LVGL_UI,_** HAS_LCD_MENU, EXTENSIBLE_UI, DWIN_CREALITY_LCD_ENHANCED, DWIN_CREALITY_LCD_JYERSUI) || BOTH(EMERGENCY_PARSER, HOST_PROMPT_SUPPORT) //01.06.2025 добавлено HAS_TFT_LVGL_UI
    #define M600_PURGE_MORE_RESUMABLE 1
  #endif
  #ifndef FILAMENT_CHANGE_SLOW_LOAD_LENGTH
    #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH 0
  #endif
#endif

And I compiled everything without errors. But the sensor hasn't worked yet. :(Printing stops immediately after it starts, as if there is no signal from the sensor.