makerbase-mks/MKS-Robin-Nano-V3.X #22: Alternative bootloader for REPRAP firmware

Lino77, 2021-04-17:
Is it possible to provide the bootloader code, which it can be adapted for the reprap firmware, or to provide a compatible one

WonderRat, 2021-04-22:
I suspect that reducing bootloader and flashing firmware at 0x8008000 will be enough.

Lino77, 2021-04-25:
240 / 5000
Übersetzungsergebnisse
the bootloader expect the firmware to be encrypted so I want an alternative bootloader that can be loaded onto the firmware without encrypting the bin file. best that you expect firmware.bin like everyone else

whhx-open, 2021-04-27:
The boot program of nano v3 has not encrypted the bin file.
The startup offset address of nano v3 is 0x800c000. I tried to modify the ldscript.ld file on rrf to adapt to nano v3, but the firmware after compiling rrf still cannot run on nano v3. I haven't found the reason yet.

Lino77, 2021-04-27:
@whhx-open 
Take a look there 
https://discord.com/channels/461605380783472640/491165528295997450/835794924330418186

https://github.com/makerbase-mks/Software/tree/master/binEncode

whhx-open, 2021-04-27:
The file encryption process is not included in the ini file, so there is no need for encryption.
[image]

Lino77, 2021-04-27:
you are right, but the board with the modified LDscript still does not boot

mks-viva, 2021-08-10:
You can refer to here: https://github.com/makerbase-mks/RepRapFirmware-for-MKS-Boards

closed 2021-11-23 by mks-viva