makerbase-mks/MKS-Robin-Nano-V3.X #23: Is possible to use J4 or J5 as controller fan?

ColdBeer72, 2021-04-18:
Board: **MKS Robin Nano-V3.X**

I use PC14 and PB1 as controlled fans for layer fan and hotend fan (in that order).

I want to use the "Controller Fan" Marlin configuration to start and stop the board fan when any driver is enabled and turns off.

The configuration block in configuration_adv.h is like this:
```
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN J4        // Set a custom pin for the controller fan
  //#define CONTROLLER_FAN_USE_Z_ONLY    // With this option only the Z axis is considered
  //#define CONTROLLER_FAN_IGNORE_Z      // Ignore Z stepper. Useful when stepper timeout is disabled.
  #define CONTROLLERFAN_SPEED_MIN      0 // (0-255) Minimum speed. (If set below this value the fan is turned off.)
  #define CONTROLLERFAN_SPEED_ACTIVE 255 // (0-255) Active speed, used when any motor is enabled
  #define CONTROLLERFAN_SPEED_IDLE     0 // (0-255) Idle speed, used when motors are disabled
  #define CONTROLLERFAN_IDLE_TIME     60 // (seconds) Extra time to keep the fan running after disabling motors
  //#define CONTROLLER_FAN_EDITABLE      // Enable M710 configurable settings
  #if ENABLED(CONTROLLER_FAN_EDITABLE)
    #define CONTROLLER_FAN_MENU          // Enable the Controller Fan submenu
  #endif
#endif
```
But with this configuration I obtain the errors:
'J4' was not declared in this scope; did you mean 'A4'?

Do I need to define it in pins file? If so: How?
May I use another connection for this objective?

mks-viva, 2021-04-19:
Hi! The motherboard's J4 is 12/24V out and it's not controller fan. Only FAN1(PC14) and FAN2(PB1) is controller fans.

ColdBeer72, 2021-04-21:
> Hi! The motherboard's J4 is 12/24V out and it's not controller fan. Only FAN1(PC14) and FAN2(PB1) is controller fans.

Robin Nano V1.2 has got just 1 PWM/TTL fan, but you still could use HE1 (PB0) as a controlled fan to connect there the hotend fan and start/stop it at some temp.

Now, with V3, we could make the same with FAN2 (PB1) (as FAN1/PC14 is used as a programmed fan for part cooling).

Can't we use HE1/PB0 on Robin Nano V3 in the same way that with V1.2?

mks-viva, 2021-04-22:
You can use HE1 as FAN3.

ColdBeer72, 2021-04-22:
> You can use HE1 as FAN3.

Ok, but, may I define HE1/PB0 as "USE_CONTROLLER_FAN" to could start/stop it based on the drivers activity?

mks-viva, 2021-04-30:
Yes you can.

closed 2021-06-25 by mks-viva