makerbase-mks/MKS-Robin-Nano-V3.X #25: Pantalla TS35 V2 no enciende / not on

Souls-Reaver, 2021-04-26:
Hola!
tengo un problema, actualice el fimware de mi robin v3 con la pantalla classic, y funciono. Luego volvi a la LVGL y ya la pantalla nunca más encendio. He probado otras firmware, los que estan aqui subidos y compilados, dejo la placa base solo con la sd y la pantalla conectada, pero sigue igual. No enciende, ni pita, ni nada de nada. me podeis ayudar?

google translate

Hi!
i have a problem, i update my robin v3 fimware with classic screen, and it works. Then he went back to the LVGL and the screen never lit up again. I have tried other firmware, those that are uploaded and compiled here, I leave the motherboard only with the sd and the screen connected, but it remains the same. It does not turn on, or beeps, or anything at all. Can you help me?

mks-viva, 2021-04-30:
Hi! You can try to use this link firmware update and check it. I think it is a firmware setting problem.
https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/tree/main/firmware/Marlin-bugfix2.0.x-MKS-2.1.2/Firmware%20bin%20for%20suport%20usb-disk
[image]

Souls-Reaver, 2021-06-09:
El problema fue que la electrónica estaba muerta.

closed 2021-06-16 by Souls-Reaver