makerbase-mks/MKS-Robin-Nano-V3.X #29: Sensorless homing

dimitribizakis, 2021-05-03:
Hi everyone, 
As i try to use sensor less homing on MKS Robin Nano v.3, i have jumpered the M2 pins and DIAG pins for X and Y.
The System is an Core XY.
HOME XY does not work so i disable it and use HOME X and HOME Y.
The problem is that the X Axis has very little sensitivity adjustment, at 80 wont move after retract and at 81 the axis bumps hard on the side.
The Y the same but it uses less sensitivity.

ColdBeer72, 2021-05-03:
I made it work, just modfiying lines (example just with X axis):
`#define X_CURRENT_HOME  X_CURRENT`
by:
`#define X_CURRENT_HOME  X_CURRENT / 2`

ColdBeer72, 2021-05-04:
> > I made it work, just modfiying lines (example just with X axis):
> > `#define X_CURRENT_HOME X_CURRENT`
> > by:
> > `#define X_CURRENT_HOME X_CURRENT / 2`
> 
> I also found out that setting up only on one motor(X) the sensitivity and the other one(Y) to 0 also fixes the problem.
> Also you have to calibrate the motor current cause making them too powerful the belts starts to slip and does not have a current peak so that the homing can me successful.

I didn't have any trouble with belts. And I readed in a Telegram group more people with the same problem on V3. It's a change Marlin (not MKS) made on last months to configuration_adv.h, because some months ago, it was defined just to the half power for homing, like I indicated.

dimitribizakis, 2021-05-07:
> I didn't have any trouble with belts. And I readed in a Telegram group more people with the same problem on V3. It's a change Marlin (not MKS) made on last months to configuration_adv.h, because [...]

Yeap, you where right, after changing #define X_CURRENT_HOME  X_CURRENT to /2 for each axis everything went smooth!
Thank you very much!

closed 2021-05-07 by dimitribizakis