makerbase-mks/MKS-Robin-Nano-V3.X #31: TS35

RunekcDK, 2021-05-03:
Hi, I have just changed my Nano 1.2 with TFT35 to a Nano V3 with TS35 but having a hard time getting the touch screen to work correctly. By default it works but for my setup I have to rotate the screen 180 degrees which is also fine but the issue is that the touch interface doesn't rotate with the screen so all the "buttons" are in the wrong place. I have tried all sorts of combinations but nothing works, is this a TS35 limitation or is it a Marlin limitation.

best regards

Rune

lilive1372, 2021-05-04:
Hi,

you just have to activate this line
#define TOUCH_ORIENTATION TOUCH_LANDSCAPE

good luck

RunekcDK, 2021-05-04:
Hi, thank you but that didn't do it, I had to send M995 to calibrate the touch screen, the old TFT35 did it after every firmware flash.

mks-viva, 2021-06-25:
Now, you can update it to the latest version and test it.

closed 2021-08-03 by mks-viva