makerbase-mks/MKS-Robin-Nano-V3.X #32: SD card access

seymon92, 2021-05-05:
Hi,
please can somebody help me? I cant acces SD card via USB, for every change I have to remove sd card. I can see SD card in Windows but it looks like it is empty. In Pronterface I get message No SD card.

Thank You

seymon92, 2021-05-24:
Yes, I have but I still cant acces SD card, I tried two cards with same result.

mks-viva, 2021-08-23:
If you need use board's TF card or Udisk function:
1. Use TF card
  - You only select **mks_robin_nano_v3** in platformio.ini file
  - You need enable **#define SDSUPPORT** in Configuration.h file

2. Use Udisk
  - You need select **mks_robin_nano_v3_usb_flash_drive** or **mks_robin_nano_v3_usb_flash_drive_msc** in platformio.ini file
  - You need enable **#define SDSUPPORT** in Configuration.h file
  - In Configuration_adv.h file: **#define USB_FLASH_DRIVE_SUPPORT**