makerbase-mks/MKS-Robin-Nano-V3.X #36: Printer stops while printing

seymon92, 2021-05-24:
I have problem with random stopping when printing, I use TM2208 (standaolne) stepper drivers with fan cooling and Mks-Robin-Nano-Marlin2.0-Firmware. It stops in a different layer and different spot.

Where could the problem be?
Thank You

mks-viva, 2021-05-28:
Hi! Can you make sure that the temperature of the heating head is constant? I guess it is caused by the fluctuation of the heating head temperature.

seymon92, 2021-06-01:
Hi, Thank You for reply. I dont think there is problem with head or bed temperature. If I do auto calibration, my printer disconnect and I cant see any information in Pronterface (but calibration continues).

I tried to print with cold head (without filament) with same result.

mks-viva, 2021-06-02:
You use USB connect to PC printing?

seymon92, 2021-06-02:
Yes,
I have not luck with this board, I cant connect RepRapDiscount Smart Controller with my board. I enable controller in Marlin and invert EXP connectors (previously I used SKR boards), but display only lights up, but shows no informations. So I print from computer.