makerbase-mks/MKS-Robin-Nano-V3.X #39: RRF firmware for the new bootloader

Lino77, 2021-06-05:
[RRF.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/6602078/RRF.zip)

Zannys, 2021-06-05:
omg, i want to test rrf in this board since i realise the wifi module has spi connections, this version of rrf i need to reflash bootloader follow this tutorial?  https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/tree/main/bootloader/No%20beep

Souls-Reaver, 2021-06-08:
Y que mejoras tiene esto?

themacboy, 2021-06-08:
excuse me! It ispossible to use RepRap in this board? Who coded that?

Zannys, 2021-06-08:
> excuse me! It ispossible to use RepRap in this board? Who coded that?

yes, i installed, but i not test yet, but is very promising
[image]

themacboy, 2021-06-08:
> 
> 
> > excuse me! It ispossible to use RepRap in this board? Who coded that?
> 
> yes, i installed, but i not test yet, but is very promising
> [image]

I will wait for a Official announcement. Thx for the effort !

mks-viva, 2021-08-10:
MKS has add bootloader for nano v3 and support rrf firmware: https://github.com/makerbase-mks/RepRapFirmware-for-MKS-Boards

closed 2021-08-16 by mks-viva