makerbase-mks/MKS-Robin-Nano-V3.X #41: Petición importante! WIFI en Colour_ui

Souls-Reaver, 2021-06-09:
Por favor, me esnecesario trabajar enviando los gcode vía WiFi a la impresora, por eso uso lvgl_ui, pero esta muy limitado este perfil respecto al colour_ui. Por eso pido, ruego más bien, que por favor, tomen las medidas necesarias para implementar esto lo antes posible. Por lo que sé, somos muchos con este problema. Un saludo.

themacboy, 2021-06-09:
I subscribe to this request !!! Bored of Cura plugins, they're useless (jono I use care! Many of us don't use it)

ColdBeer72, 2021-06-14:
I subscribe too!!!

mks-viva, 2021-06-25:
@Souls-Reaver @themacboy @ColdBeer72 I will feed this demand back to the engineer.

Souls-Reaver, 2021-06-25:
> @ Souls-Reaver @themacboy @ ColdBeer72 Enviaré esta demanda al ingeniero.

Genial, muchas gracias m, de verdad

Souls-Reaver, 2021-07-09:
Se sabe algo, sobre este tema?

trashypopashy, 2021-10-19:
To consider does not completely mean, to do so. Will there be support after all the months of no reaction in this thread?

trashypopashy, 2021-11-10:
When will the development be done?

Mihalch999, 2022-01-20:
Can you please tell me when WIFI support will appear in the COLOR UI interface?