makerbase-mks/MKS-Robin-Nano-V3.X #43: Flashing the bootloader using OpenOCD

dzervas, 2021-06-19:
The board is completely dead to me after trying to flash Klipper through the SD. I can't see anything on USB.
I've attached ST-Link v2 and trying to flash the `Beep` firmware through OpenOCD:

```
openocd -f interface/stlink.cfg -c "transport select hla_swd" -f target/stm32f4x.cfg
reset halt
program /home/dzervas/Downloads/mks\ robin\ v3.0/nano_v3_bootloader.bin 0x8000000 verify
```

But nothing changes.
Any pointers? Maybe the offset it incorrect? Maybe the bootloader is encrypted? (#22 ) I can't find any alternative bootloader (stm32 DFU maybe? or HID?)
@whhx-open you seem to know your way around the board

dzervas, 2021-06-19:
Also I found that:

```
Sections 0 and 1 : (0x08000000 -> 0x08007FFF) : Bootloader
Sections 2 and 3 : (0x08008000 -> 0x0800FFFF) : Non-volatile memory
Sections 4 through 11 : (0x08010000 -> 0x080FFFFFF) : Application program.
```

Here: https://community.arm.com/developer/tools-software/tools/f/keil-forum/30625/stm32f4-bootloader-data-flash-and-application

If that's true, I may have corrupted the NVM so the bootloader crashes. Maybe I need to fix that?

mks-viva, 2021-06-25:
You can use J-link connect Nano V3's SWD interface

dzervas, 2021-06-25:
I already have connected to the SWD interface as I describe above, the question is what's the correct offset to flash the bootloader and the actual running firmware

iNDeX999, 2021-06-27:
Same Problem here,

i try to flash Klipper via USB Stick. But the FW update stops at 100% with beep.
Nothing more happend.

After power off/on the USB is dead / tft dark

dzervas, 2021-06-28:
this needs a deep dive into the MCU's datasheet as apparently the devs have no idea what we're talking about and I'm not sure that I have the time to do that. If you have any lead, leave it here, just in case

mks-viva, 2021-06-29:
MKS Robin Nano address
BootLoader：0x08000000-0x0800C000
App start address：0x0800C000

iNDeX999, 2021-06-29:
@dzervas did you get it to work?

dzervas, 2021-06-30:
didn't have time to look into it yet - maybe on the weekend I'll give it a go

tiagooliveira95, 2021-07-16:
@dzervas were you able to fix it via openocd?  my brand new board doesn't seem to work at all, I tried to program marlin and the bootloader but nothing happens.
I'm also unable to detect the board from my pc

baiak, 2021-07-18:
> 
> 
> @dzervas were you able to fix it via openocd? my brand new board doesn't seem to work at all, I tried to program marlin and the bootloader but nothing happens.
> I'm also unable to detect the board from my pc

It happened the same with me, I`m unabled to update the firmware... in platform.io I search for devices, the COM port where board is connected appears in the list, but nothing happen when I try to update.

tiagooliveira95, 2021-07-20:
@dzervas @baiak I was able to fix the board via SWD using my Pi 4

# https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/issues/48#issuecomment-883009613

dzervas, 2021-07-20:
@tiagooliveira95 does the device something "special" to come alive? does it need 12v or 24v or should it be able to boot just from USB power? Do you see the device as a USB peripheral?

dzervas, 2021-07-20:
OMG it worked. I then flash the "bootloader" files via the SDCard and it worked
@tiagooliveira95 you're the best, if you have a ko-fi.com or such, drop me a link to buy you a beer or smth

closed 2021-07-20 by dzervas

LIAS81, 2022-04-11:
> OMG it worked. I then flash the "bootloader" files via the SDCard and it worked @tiagooliveira95 you're the best, if you have a ko-fi.com or such, drop me a link to buy you a beer or smth

Hi, my board is dead, got an st-link v2 device but I can't connect, Did you fixed it with raspberry or st-link v2?

LIAS81, 2022-04-12:
Just for the record if anyone experience the same situation, I was able to fix it with stlink v2 and stlink cube programmer running on ubuntu linux 22.04,  I tried first on windows without success, tried with openocd running on linux without success because openocd have an issue with linux newest USB library, cube programmer on linux worked like a charm.

3.3v drive IC power jumper was selected on card and 3.3V power and ground was provided through st link device  DIO,CLK, RST cables connected.

Cube programmer connection settings:
Port: SWD
Frequency: 950
Mode: Normal
Acces Port: 0
Reset Mode: Software reset
Speed: Reliable
Shared: Disabled

After connection to the board go to Erasing & Programming section, browse for the bootloader file, make sure that starting address is 0x08000000 and click "Start Programming" button

After bootloader is downloaded to the card, firmware upgrade can be done via USB or SD card 

I want to thank @thiagooliveira95 and @dzervas for posting their solutions because they shed some light on my way to fix my board,