makerbase-mks/MKS-Robin-Nano-V3.X #45: SD file transfer does not work - MKS Robin Nano V3

pagabee, 2021-06-28:
The SD file transfer does not work neither USB nor Wifi.
After starting the transfer, _"No firmware response detected for sd card action. Assuming success and continue"_ message appears and the temperature log stops.
I tried out more SD cards and USB sticks.

This files were used:
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/master/Firmware

Can anybody help me?

[image]
[image]

pagabee, 2021-07-11:
The file upload works via USB, if the G-code flavour is Cura in Slic3r.

[image]

pagabee, 2021-07-11:
Same issue with Pronterface, file upload is ok through USB, but the Wifi transfer is nok.
The screenn shoes: Please check wether memory device insert!

[image]

pagabee, 2021-08-31:
Is there an other way to upload files through wifi besides Cura?

As I see the wifi modul has web view interface.
Is it supported?

[image]

Tompudex, 2021-09-20:
> 
> 
> Is there an other way to upload files through wifi besides Cura?
> 
> As I see the wifi modul has web view interface.
> Is it supported?
> 
> [image]

nem. erre még várni kell. egye
löre a wifi fw-t tudod itt cserélni. meg beállítani a csatlakozását.

pagabee, 2021-09-29:
I tried out the new ESP3D 3.0 firmware on MKS WIFI.
The file transfer via wifi to SD card or U drive does not work, and communication stops after that.

[image]

[image]

[image]

pagabee, 2021-09-29:
There is no problem with Cura + MKS Wifi Plugin, but I am not using Cura slicer.