makerbase-mks/MKS-Robin-Nano-V3.X #47: TMC5160

Sangra69, 2021-07-06:
Has anyone gotten this stepper driver to work with this board? When I set it to SPI mode I get a driver error. Then it says printer halted.

mks-viva, 2021-07-07:
You need set it:
1. In Configuration.h file, define DRIVER_TYPE to TMC5160, for example #define X_DRIVER_TYPE  TMC5160
2. In Configuration_adv.h file, enable #define TMC_USE_SW_SPI
3. You board need use jumpers set to SPI mode 
[image]
4. You need use dc12-24v power

Sangra69, 2021-07-07:
Dang I did do all that with E0. My extruder had no response. I am going to have to with a new configuration. H file and try again.

Sangra69, 2021-07-08:
Just letting you guys know that the BigTreeTech v1.2 TMC5160 didn't work at all. I wonder if Makerbase has a 5160 stepper driver?

mks-viva, 2021-07-19:
We do not produce TMC5160, maybe you need to try, does TMC5160 need flying lead, such as flying lead to set SPI mode...

closed 2021-08-04 by mks-viva

vogon, 2022-08-08:
not sure if you're still having this issue, but in the off chance you are, https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/issues/2 and the schematics in this repo pointed me at the solution -- out of the box, the CLK pin of the TMC5160 isn't tied to a clock signal on the robin nano 3.x, so bending the CLK pin back and letting it be tied to ground by the pull-down resistor (thus clocking the TMC5160 internally) fixed the problems I was having with getting it to respond.