makerbase-mks/MKS-Robin-Nano-V3.X #49: Bed / Nozzle Temps Reading Random Values

CollinCBoring, 2021-07-18:
Problem: I have fully wired in the Robin Nano V3 and I am getting temp readings that bounce between -8000 and 8000.  The values appear to be random and change quickly.  When I unplug the 100K thermistors from the board, the values go to zero. No other issues are present with controlling the movement of the machine, homing, or auto-bed leveling.

Tests I have Completed: 
1. Verified the 100K thermistor is functional and tried various new sensors.
2. Tested the firmware across other Robin Nano V3 boards I have.

Files Attached:
1. Marlin Firmware 2.0 from the Current link folder. https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware
2. Pictures from my screen.

Configured for:
1. H43 Touchscreen
2. Sensorless Homing on X & Y
3. BL Touch
4. TMC2209 in UART Mode
5. Smart Filament Runout Sensor

[Marlin Firmware.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/6835619/Marlin.Firmware.zip)
[image]
[image]
[image]
[image]
[image]
[image]

mks-viva, 2021-07-19:
I will report this problem to the engineer, can you try to use the latest marlin firmware?

CollinCBoring, 2021-07-19:
> I will report this problem to the engineer, can you try to use the latest marlin firmware?

Yes I will download the latest firmware today and see if it changes the readings. I will post the results within the next 24 hours.

CollinCBoring, 2021-07-19:
I downloaded the latest version of Marlin 2,0 from the Makerbase Repository and I have the same issue. I did learn though when I configure the board for the TFT35 V2.0, the temp readings are normal. I tried flashing the H43 with NO BEEP from H43_Firmware_for_Malin and H43_Firmware_1.31, same issue with both.

Here are the steps I take to create the problem.
1. Download the latest version of Marlin 2.0 from the makerbase repository.
2.  Disable the following:
//#define MKS_TS35_V2_0
//#define TFT_LVGL_UI
//#define TOUCH_SCREEN
3. Enable the following:
#define DGUS_LCD_UI_MKS
4. Compile and load to Robin Nano V3 using flash drive
5. Update the H43 using the latest DWIN_SET.

CollinCBoring, 2021-07-22:
Any progress?

CollinCBoring, 2021-08-03:
Any Progress @makerbase-mks @MKS-Sean

Grant09, 2021-08-11:
hi, we have the same issue, pls help

slapukas, 2021-08-28:
hi, we have the same issue, pls help

Grant09, 2021-08-28:
> hi, we have the same issue, pls help

Hey man send me a email at grant@cinch-it.co.za and I'll send you the files to fix jt

CollinCBoring, 2021-08-30:
@MKS-Sean @makerbase-mks Any Progress Here???