makerbase-mks/MKS-Robin-Nano-V3.X #55: M503 command have only OK response through Wifi connection

pagabee, 2021-09-06:
I wanted to change the eeprom settings in Repetier-Host through Wifi connection, but it is not possible, the window is empty. The M503 command has only OK respose.

[image]

M115 and M122 commands have the same behaviours:

Wifi connection:
[image]
[image]
[image]

USB connection:
[image]
[image]
[image]

mks-viva, 2021-09-13:
If you use pronterface connect it, send M503 will return what?

pagabee, 2021-09-13:
The same as with PuTTY:

[image]
[image]

mtsegundo, 2023-10-09:
Did you find a way to fix this problem?