makerbase-mks/MKS-Robin-Nano-V3.X #56: POWER_LOSS_RECOVERY not working properly MKS Robin nano V3

riff-electronics, 2021-09-13:
I have tried anything but always not working properly on marlin 2.0.9.1. When  power outage occure Z axis not lift up. This caused failed on when power on again. Please MKS help fix this trouble. Thanks

mks-viva, 2021-09-15:
Please you marlin source code to me.

riff-electronics, 2021-09-16:
https://github.com/riff-electronics/Marlin-2.0.9.1-coreXY

riff-electronics, 2021-09-16:
https://drive.google.com/file/d/1Ks-J7AWKY-qRfBW2iSeLpPUwtGFM8_--/view?usp=sharing

riff-electronics, 2021-10-30:
Still not yet work, when power outage z axis raising up, but after power on again, choise for resume print or stop print not appear, direct to common display. I think like machine not reading PLR file on cd card or USB

andelsalam, 2022-02-03:
l have same problem with marlin 2.x bugfix any idea ,,???

Khajeh0homa, 2025-01-20:
and me. l have same problem with mks robin nano v3 & marlin 2.x bugfix