makerbase-mks/MKS-Robin-Nano-V3.X #60: Robin Nano V3 rev 004 + display H43 + USB stick NOT start

spammchik, 2021-09-28:
Configuration: Robin Nano V3 rev 004 + display H43 V1.0 or V1.1 + USB stick NOT start
**Problem:** If you connect a USB drive to the Robin Nano V3 rev 004 and turn on the power via USB or 12V, **the electronics will not boot, display not work.**
Everything works fine without a USB stick.
If you plug in the USB stick after the electronics turn on, then everything works fine.
If there is no display H43, then the electronics starts with a USB drive fine.
This problem is not present on Robin Nano V3 rev 003.
**Question:** How to make the electronics Robin Nano V3 rev 004 + display H43 V1.0 or V1.1  boot and work with the USB drive connected?

mks-viva, 2021-09-29:
Hi! I want to know, you use same the firmware on 003 and 004?

spammchik, 2021-09-29:
The firmware is the same for 003 and 004.

mks-viva, 2021-09-30:
004 hardware changes
[image]

closed 2021-11-29 by spammchik

KORSiRO, 2021-12-05:
Hi look this : https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/issues/76
I managed to find the solution after hours of testing;)