makerbase-mks/MKS-Robin-Nano-V3.X #62: SD Only = reboot after heating | USB+SD = Both not recognized | UBS cable = success

ImmortalTeknik, 2021-10-02:
Hello guys,

I couldn't find anything relating to my problem so far on web so here I am asking your help.

**Hardware:**
- MKS Robin Nano V3 (Marlin from MKS flash drive msc)
- TFT 35 MKS (Color UI)
- MKS Wifi
- TMC2209
- 12V PSU
- Prusa I3 
- The printer was working perfectly on a 8 Bit MKS Base 1.4 for 6 years.
- San Disk 2GB SD
- Kingston 16GB USB

**The issue:**
- With SD only enable in the config, the sd card is recognized, I launch a job, then marlin reboot as soon as the temps are 
  reached (190°C and 40°C). On 1 trial to 10, printer has time to home before it reboot when I relaunch the job very quicly before 
  temps drop too much.
- With USB Flash Drive Support and Multi Volume, neither UBS stick or SD card is recognized.
- Printing from USB cable work perfectly.
- ESP3D can control printer but crash when trying to write on SD Card.

What I tried:
-Remove MKS Wifi
-exFat or Fat32
-Enable Disable OTG feature
-Flash drive w/without msc.
-SPI speed half

Attached are my Cfg files.

It's my first post here.
I was so happy to swap my old 8 bit board for a modern hardware but I am now totally disapointed. 8 bit was limited but simple and less prone to bug I think, 32 bit seems to be far more complex and less reliable. Hope I am wrong, we will see.

[Configuration.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/7271300/Configuration.txt)
[Configuration_adv.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/7271301/Configuration_adv.txt)
..

mks-viva, 2021-10-07:
Hi! What is you set in platfromio.ini file env=?
You can refer to here set:
[image]

ImmortalTeknik, 2021-10-07:
Hello Viva,

Thanks for your reply.

I tried with both _usb_flash_drive and _usb_flash_drive_msc.

ImmortalTeknik, 2021-10-13:
Nice support from MKS...

ImmortalTeknik, 2021-10-13:
So, after days and nights of head sracthing, I finally solved my issue: my gcode starting script was making marlin to crash.
I just removed it and write a new one (default cura).

closed 2021-10-13 by ImmortalTeknik