makerbase-mks/MKS-Robin-Nano-V3.X #72: Board reboot with any USB stick

ImmortalTeknik, 2021-10-31:
Under LVGL UI with lastest firmware version, as soon as I plug any USB stick, the board freeze and reboot, so I can't print from USB.

For information, I use:
mks_robin_nano_v3_usb_flash_drive
#define USB_FLASH_DRIVE_SUPPORT
#define MULTI_VOLUME

SD card is read, but not USB stick. I tried with 5 USB sticks.

What did I miss ?

mks-viva, 2021-11-05:
What firmware version you using?

btnkalyoncu, 2022-03-18:
I have the same issue. Could you tell me how you solved it?

Ertil-Noorison, 2022-07-13:
I use Mks-Robin-Nano-Marlin2.0-Firmware. CONFIGURATION_H_VERSION 02000901
Board Robin-Nano-V3.1 + TS35-V2.
USB and SD work well.
USB and SD work well.
in the firmware, set version from "mks_robin_nano_v3_usb_flash_drive"