makerbase-mks/MKS-Robin-Nano-V3.X #73: No BLtouch settings menu

ImmortalTeknik, 2021-10-31:
Hi,

I setted up my BL touch on my Robin Nano V3 following the MKS tutorial but one thing is missing, the BLtouch setting menu where you can also set the Z probe offset.

How to enable this menu ?

zMRCASTLEz, 2021-11-10:
The "AutoLevel" option is supposed to replace the "Level" option in the tools menu, but despite following the entire video I don't get that option.

marcelobsilva18, 2021-11-17:
You have to choose  #define AUTO_BED_LEVELING_BILINEAR in order for it to change. I managed to get there but the screen doesn't jump to auto leveling mode even thought it does it. The problem is it doesn't allow me to set the offset. Any ideas?

zMRCASTLEz, 2021-11-18:
Thank you very much, now the autolevel menu appears and I have the same problem as you, I guess it will be a bug. What I did to be able to set my z offset was the following:
1.- Deactivate "#define MIN_SOFTWARE_ENDSTOP_Z" (This will allow us to lower the z axis below 0)
[image]

2.- Then with this parameter deactivated we compile and update the printer firmware.
3.- We do autohome.
4.- Then from the printer we go to the tools menu and then to the move section.
5.-Then we lower the z axis little by little until the nozzle just touches our sheet of paper.
6.- We add the value where our nozzle touched the sheet of paper to the value that our z offset already had, for example if our z offset was at -1.0 and at -1.2 the nozzle touched our hot bed, our final offset. will be -2.2.
7.- Once we have our final offset we go to "Settings-> config-> Leveling Settings-> Nozzle-to-probe offset settings" and enter our value.
8.- Finally we print a level test.
I hope this momentary solution will help you to level your bed until this bug is resolved.
(I'm Chilean and I don't speak English well enough, so I apologize in case of spelling errors)

marcelobsilva18, 2021-11-18:
Thank you, I will try it! I think it is a good solution for now. Your English is good, never appolagise for not speaking a second language perfectly! (I an Portuguese, not original English speaker either!)

Jcm0724, 2021-12-10:
I've exactly the same issue even define AUTO_BED_LEVELING_BILINEAR. No autolevel setting... I've done everything you've done but still nothing..

Smarthelpers, 2022-02-07:
Any news? It would be wonderful if MKS would provide us a solution for this problem.

SebastianJuch, 2022-02-10:
Hi there, at the moment I‘m in contact with the support over Facebook. They are showing me configuration‘s which are not published in GitHub and also not working. I will try to get here a solution but it will take some time because of the time difference between china and germany. I hope we will get a solution. Hopefully. 😂👍

MKS-hosiey, 2022-02-11:
> Hello, is there no z_offset button in the menu after auto-leveling? The latest firmware blocks it, we'll add it

Smarthelpers, 2022-02-28:
Any news about this topic?

SebastianJuch, 2022-02-28:
No, last message from 19.02.2022

„ Sorry, there is a problem with our modification, there are many problems in the modification test, so it will not be released, and the firmware will take a while.“

No updates until now!

Sorry for the bad news. But you can use the Marlin Ui. Their Wizard is working 😉

Smarthelpers, 2022-02-28:
Thanks for the reply. So all we can do is to wait for the UI-Update.

rwoldberg, 2022-03-07:
Any update on this?

dendroca, 2022-03-31:
Did you have update ?  I think i will ask for refund and go with bigtreetech!!!
Too much problem with mks and no enought  files and documentation. I m really up set and have lost too many hours to try to fix this issu!

rwoldberg, 2022-03-31:
I started with the marlin bug fix 2.0.x branch from their github and then updated that.  I copied the bltouch dialog from a previous commit from the MKS github.  I might create a pull request into the marlin branch but it is not ready just yet.

zMRCASTLEz, 2022-04-02:
The best solution to this problem is to buy a bigtreetech motherboard, it is not acceptable that for months we have no solution. I have been very disappointed with the customer service from MKS and I do not plan to buy any product from this brand again.
I bought a skr mini e3 v3 for my ender 3 and it has saved me from having to keep wasting my time trying to solve this problem.