makerbase-mks/MKS-Robin-Nano-V3.X #75: Cannot compile #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled"

Bergateo, 2021-11-06:
This error, #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled", is displayed when I try to compile and for this error I cannot create the .bin file to upload.

Anyone has any clue to solve this error?
My setup has:
- h43 screen
- TMC2208 in uart mode

Bergateo, 2021-11-06:
Managed to solve it, it was compiling indeed but somehow the board didn't get the update

closed 2021-11-06 by Bergateo

jessepietila, 2023-03-17:
I get the warning as well---> "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled.

What to do?

Romain80200, 2023-03-27:
Hello, you can ad that in configuration.h, line66:

//#define CUSTOM_VERSION_FILE Version.h // Path from the root directory (no quotes)
#define DIAG_JUMPERS_REMOVED