makerbase-mks/MKS-Robin-Nano-V3.X #76: MKS ROBIN NANO V work with H43

KORSiRO, 2021-12-05:


KORSiRO, 2021-12-05:
Hello everyone !

After several tests and trials I finally found the right configuration for everything to work:

- H43 OK
- USB Storage OK
- USB connection (PC) OK

**_1. H43 :_**

default_envs = mks_robin_nano_v3_usb_flash_drive_msc

#define SERIAL_PORT -1
#define BAUDRATE 250000
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
#define DGUS_LCD_UI_MKS

//#define TFT_LVGL_UI
//#define TOUCH_SCREEN
//#define MKS_TS35_V2_0
//#define DISPLAY_CHARSET_HD44780 JAPANESE

**_2. USB STORAGE :_**

#define SDSUPPORT

#define USB_FLASH_DRIVE_SUPPORT

#define USE_OTG_USB_HOST

#define MULTI_VOLUME
  #if ENABLED(MULTI_VOLUME)
    #define VOLUME_SD_ONBOARD       1
    #define VOLUME_USB_FLASH_DRIVE  2
    #define DEFAULT_VOLUME SD_ONBOARD
    #define DEFAULT_SHARED_VOLUME VOLUME_USB_FLASH_DRIVE
  #endif

3. USB connextion PC :

#define SERIAL_PORT_2 2
#define BAUDRATE_2 115200

I hope this makes things easier for you ;)

CLP58, 2022-07-01:
Can you tell me if I can't use WIFI anymore with the H43 on the nano v3?