makerbase-mks/MKS-Robin-Nano-V3.X #78: Cannot create generic_device.c.o:

Tibu11, 2021-12-10:
Hello I'm a total noob can anyone tell me how to fix this? 
 
> Executing task: C:\Users\Kakkarot\.platformio\penv\Scripts\platformio.exe run <

Processing mks_robin_nano_v3_usb_flash_drive (platform: ststm32@~12.1; board: marlin_STM32F407VGT6_CCM; framework: arduino)
------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Copying MKS Assets
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32F407VGT6_CCM.html
PLATFORM: ST STM32 (12.1.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 0.0.0
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <lvgl> 6.1.1
|-- <TMCStepper> 0.7.3
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\Print.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\USB.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\VirtIOSerial.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\WString.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\abi.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\board.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\hooks.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\itoa.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\main.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\new.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\stm32\OpenAMP\libmetal\device.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\stm32\OpenAMP\libmetal\generic\condition.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\stm32\OpenAMP\libmetal\generic\cortexm\sys.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_device.c.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_init.c.o

Assembler messages:
Fatal error: can't create .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_device.c.o: No such file or directory
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_io.c.o
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_device.c.o] Error 1
========================================== [FAILED] Took 17.24 seconds ==========================================

Environment                        Status    Duration
---------------------------------  --------  ------------
mks_robin_nano_v3_usb_flash_drive  FAILED    00:00:17.242
===================================== 1 failed, 0 succeeded in 00:00:17.242 ===================================== 
The terminal process "C:\Users\Kakkarot\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

mks-viva, 2021-12-13:
Provide your source code to me.

Tibu11, 2021-12-14:
It's makerbase's source code, this is the error I get when I try compiling before making any changes. https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X

mks-viva, 2021-12-17:
You can use this link firmware: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware.
The firmware under the path of MKS Robin Nano V3.X is 10 months ago. Maybe the Vscode compilation environment has changed, causing compilation errors.