makerbase-mks/MKS-Robin-Nano-V3.X #81: Soft Brick when updating bootloader on Nano V3.1

NotAFood, 2022-01-08:
Hi. I recently changed my bootloader from beep to no beep on my V3.1 I copied all 3 of the files onto the SD card and the update completed; however, now it is stuck in a boot loop and turns the display on and off while beeping. I can change the bootloader back to the beeping version and it updates along with changing the file to .CUR but it will not read or apply any Robin_nano_v3.bin file on the card. Regardless of which one is used, the device keeps turning on and off and will not accept new Marlin firmware on the SD card

MKS-hosiey, 2022-01-10:
Hi, have you tried changing the SD card to update the firmware?

NotAFood, 2022-01-11:
Yes. I've tried two of them with the same result. I know it is reading the card because it will accept a boot loader file from it.

NotAFood, 2022-03-22:
I bought an ST Link on Amazon and was able to flash the board over the SWD port using STM32 cube programmer. Use 0x0800C000 as the address to flash to.

mavant, 2022-06-25:
@NotAFood, can you say more about how you connected to the board? Were there any jumpers that needed to be set?
I'm currently seeing STM32cubeprogrammer spin forever in the 'connecting' state, although the board LEDs do turn on as soon as the ST-link dongle is physically connnected.

NotAFood, 2022-06-25:
> @NotAFood, can you say more about how you connected to the board? Were there any jumpers that needed to be set?
> 
> I'm currently seeing STM32cubeprogrammer spin forever in the 'connecting' state, although the board LEDs do turn on as soon as the ST-link dongle is physically connnected.

I always had it hooked up via external 12/24v power. Make sure to set the connection speed to 1800 instead of 4000. If normal mode doesn't work, I've also had success with "under reset". Didn't recall having to set a jumper.

mavant, 2022-06-30:
I never quite got STM32cubeprogrammer to connect, but the older "ST-Link Tool" connected immediately on the first attempt and I was able to flash with any speed between 900 and 4000 Mhz (I didn't try any lower). I'm not sure why this would be, but I have been supplying power via the 3.3v pin on the st-link instead of via the 12/24v screw block; maybe that's why.

I still seem to be doing something wrong, though; I can't get a firmware update to be applied from the TF card. Sorry if this is a dumb question, but - am I correct in thinking that you flashed 'nano_v3_bootloader.bin' at 0x08000000 and 'firmware.bin' at 0x0800C000?
When I do that (with the no-beep firmware, starting from an otherwise clean-wiped board), I end up with these checksums:
```
23:41:39 : [firmware.bin] opened successfully.
23:41:39 : [firmware.bin] checksum : 0x0242B529 
23:43:07 : Memory programmed in 10s and 468ms.
23:43:07 : Verification...OK
23:43:07 : Programmed memory Checksum: 0x0242B529
23:43:08 : Flash memory [0x08000000:0x08080000] Checksum: 0x0475091D
```

but then update from SD still fails, so I think I must be flashing the wrong binary.

cyh924417122, 2022-07-25:
> I bought an ST Link on Amazon and was able to flash the board over the SWD port using STM32 cube programmer. Use 0x0800C000 as the address to flash to.

hello, witch hex you used?My Board's MCU have been changed and the firmware.hex flash in could not start work

abmantis, 2022-07-29:
My Robin Nano 3.1 also got bricked after trying to flash the non-beep bootloader! 
I was able to flash back the beep bootloader using the SWD port, but I still cannot get it to flash the firmware using the sdcard. 
Are the bootloader files updated for the 3.1?!

abmantis, 2022-07-30:
@mks-viva help? Maybe MKS needs to update the bootloader binaries on github for the v3.1?

closed 2022-12-29 by NotAFood

NotAFood, 2023-03-22:
Yeah I eventually had the problem too. I couldn’t get it to work again and ultimately went with the more user friendly Fystec S6

MKS-hosiey, 2023-03-23:
> after a short circuit on my board, I tried to flash it but it only shows the message that the memory is blocked, it is not possible to read or write, ask would it be the processor burned or the bios?

You can use a multimeter to measure the capacitance next to the main chip, if there is a short circuit, then the main board is burnt out