makerbase-mks/MKS-Robin-Nano-V3.X #82: MKS H43 display not working when enabeling E1 stepper driver

Michi1901, 2022-01-08:
Hello, the display was working fine until i enabled dual z axis support and the display was not able to communicate with my Nano v3.0 anymore. The H43 display is connected over the UART 3 interface. I am using E1 stepper for second Z axis. Is this expected behavior? Could I maybe use a different UART for the display? UART 1 from WIFI is free but was not able to get it working.

Michi1901, 2022-01-09:
Was able to get it resolved. In the end it was my PT1000 temperature sensor at fault. The display seems to freeze when there is nothing connected on the TH1 port on the motherboard and the value 1047 is set in Marlin.

closed 2022-01-09 by Michi1901

ysnuysl, 2023-12-12:
Hello,

How were you able to set up the 2nd motor and driver correctly for the Z? I also get "error: #error "No E stepper plug left for Z2! #95" error.