makerbase-mks/MKS-Robin-Nano-V3.X #85: Bobin Nano V3,screen problem

sima35, 2022-01-29:
I'm making a coreXY machine.
I'm rewriting Marlin's data from you.
I transcribe TMC2900,3D Touch, MKS PWC.
On Nano v2, everything worked.
But when I load the Firmware, everything is ugly on
 the tft display and it restarts after a while.
But when I uploaded your firmware for Sapphire pro, it worked properly only my parameters are not there.
I would need professional help
I give in doc. just my basic settings.
Thank you in advance for your help

[Configuration adv.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/7964711/Configuration.adv.txt)
[Configuration h.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/7964712/Configuration.h.txt)

[image]

[image]

rick090980, 2022-02-07:
Unfortunately don't have anything to offer but I experienced the same problem on my Longer LK4 Pro. Until I find a fix I decided to use the Marlin "color ui" instead of "lvgl ui". I think I like it better.

mykepredko, 2022-02-07:
Hey @sima35 & @rick090980  

I'm using "TFT_LVGL_UI" and the screen is working fine.  

I do see the random reboots as well as poor touch screen response and reboots while it's being touched.  

Hopefully somebody who's worked on this can comment.

Thanx.

Smarthelpers, 2022-02-08:
I had the same issue, but now it is running well after I wasted many days for try and error. I used the last available Marlin Version from the MKS-Github and modified the configuration files. Only I can't get any Z-Offset menue. Do you have Z-Offset-Options after Bed Leveling? MKS showing this on YouTube (https://youtu.be/YhSYDavYRYs?t=303), but I can't get this options. I sent MKS a message but they did not gave me a helpful answer. It seems like they only making marketing with functionality who not really exists.

mykepredko, 2022-02-09:
@sima35 & @rick090980 

I just followed the advice of @Smarthelpers and started with the latest bugfix version of Marlin and changed the platformio.ini, Configuration.h & Configuration_adv.h files to match the settings in the MKS version of Marlin.  

I've included these files in this post so you don't have to go through the same exercise of doing a kdiff on the Robin V3 files versus the Marlin bugfix and trying to decide what the different #define labels and their values should be.  I should point out that the values are for an Ender 3 (more or less) so you will have to change the operating parameters for your printer.  Having said that, I do have my steppers running, although not in the right directions.  

The problems I had with the display have gone away.  Now, it's a really sh!tty UI and it doesn't always respond correctly, the scroll wheel doesn't go everywhere and I found I needed a touch screen stylus for the smaller on screen controls but everything  displays properly and the Nano V3 doesn't reset periodically.  

Let me know how things work for you.

myke

[Configuration & platformio files.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/files/8029522/Configuration.platformio.files.zip)

rwoldberg, 2022-03-06:
I have this same issue.  A fix would be nice.

Mihan1988, 2022-03-08:
I have the same problem.

awakenedlast, 2022-03-28:
I have this problem whenever I send a print from Cura using the MKS wifi. The screen looks exactly like this.

ikkez, 2022-04-05:
> I have this problem whenever I send a print from Cura using the MKS wifi. The screen looks exactly like this.

I can confirm this. same problem here when sending data from cura via wifi.

Smarthelpers, 2022-04-05:
Unfortunately MKS didn't give any reaction. I wrote them on Aliexpress, but all I got was a message that I have to wait for an update. This was many weeks ago...  It seems like that they only want sell boards and that it is our problem if the firmware is not running. I bought a lot of boards from MKS and nearly all have firmware issues.

naxneri, 2022-04-08:
in the platformio.ini file change default_envs to 

default_envs = mks_robin_nano_v3_usb_flash_drive_msc

that will solve the problem

sima35, 2022-05-24:
> v súbore platformio.ini zmeňte default_envs na
> 
> default_envs = mks_robin_nano_v3_usb_flash_drive_msc
> 
> tým sa problém vyrieši

I have set it right from the beginning.

sima35, 2022-05-24:
> > v súbore platformio.ini zmeňte default_envs na
> > default_envs = mks_robin_nano_v3_usb_flash_drive_msc
> > tým sa problém vyrieši
> 
> I have set it right from the beginning.

the problem remains the same

fatih5228, 2023-12-22:
For display problem, connect Ferrite Core to EXT1 output.
https://www.amazon.com/Ferrite-Clamp-Cores-Round-Filter/dp/B005T8UQAM/ref=sr_1_14?crid=3TUDQXEN3TLCO&keywords=Ferrite+Core&qid=1703252127&sprefix=ferrite+core%2Caps%2C211&sr=8-14
aliexpress.com/i/4000803662446.html?gatewayAdapt=glo2tur