makerbase-mks/MKS-Robin-Nano-V3.X #92: MKS ts 35V2  on Robin Nano v3 No Offset wizzard Menu

Gby33, 2022-02-20:
As suggested in the title no offset wizzard appears using TFT_LVGL_UI. 
Although it appears on an MKS vid https://www.youtube.com/watch?v=YhSYDavYRYs&t=303s
I use the latest version of Marlin to build my project.
Does it means that this fonctionality is not implemented in this version ?

rwoldberg, 2022-03-07:
I would like to know how to fix this issue as well

awakenedlast, 2022-03-28:
This is still a problem. Using the latest firmware the z-probe offset menu is not accessible, nor does it appear after running autolevel. The nozzle just stays where it's at once autolevel is complete.

ikkez, 2022-03-31:
you can enable BABYSTEPS and use that menu while printing the first layer. the setting is automatically stored and applied on further prints then.

awakenedlast, 2022-03-31:
> you can enable BABYSTEPS (which sets z-probe offset) and use that menu while printing the first layer. the setting is automatically stored and applied on further prints then.

Yes, this has been my workaround too. Unfortunately, it still does not solve the issue of the z offset screen not being shown outside of a print, which is the posted issue here.

rwoldberg, 2022-03-31:
I tracked down the issue to this merge, https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/commit/c03dfaf0c0185434a6d87cd13c24935f1cfbaead, draw_bltouch_settings.cpp was deleted.  I'm not sure but I think that might have been a bad merge.

Plizze, 2022-05-14:
This display can work with MKS Gen v.14?