makerbase-mks/MKS-Robin-Nano-V3.X #96: Navigator not working with TFT_COLOR_UI

MoustafaElkady, 2022-03-04:
Hello,
I'm using MKS nano robin v3 with MKS_TS35_V2_0
the problem is, when i use TFT_COLOR_UI , the navigator botton is not working
and when i use TFT_LVGL_UI it shows green screen

i feel like its a wrong decision to get this board

MKS-hosiey, 2022-03-05:
Green screen with lvgl ui? you have updated assets image file

Mihan1988, 2022-03-08:
I have the same problem.

MKS-hosiey, 2022-03-08:
Can you upload a picture of the green screen interface?

MoustafaElkady, 2022-03-08:
i used the branch https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/Marlin-bugfix2.0.x-MKS-2.1.3 and it fixed

Mihan1988, 2022-03-08:
> Можете ли вы загрузить изображение интерфейса зеленого экрана?
[image]

MKS-hosiey, 2022-03-08:
Hi, is your cable bent?

MKS-hosiey, 2022-03-08:
Or turn on the rotary encoder?

MKS-hosiey, 2022-03-08:
Please disable it in pins_robin_nano_V3.h
   #define LCD_USE_DMA_SPI

Mihan1988, 2022-03-08:
> я использовал ветку https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/Marlin-bugfix2.0.x-MKS-2.1.3 и исправил

Thanks. Your firmware helped.

Thoran71, 2022-03-10:
Hello, 
This Fix works fine . Now i have the problem with no USB Support. 
how can i fix this problem ?

MoustafaElkady, 2022-03-17:
> Hello, This Fix works fine . Now i have the problem with no USB Support. how can i fix this problem ?

same here