makerbase-mks/MKS-Robin-Nano-V3.X #100: Print-Overview Screen causes artifacts

ikkez, 2022-04-04:
Hi MSK team. I was investigating a problem that I was looking into for several months now. I'm having Blobs and Zits on my print walls and tried everything to find the issue, swaps Nozzles, Filament and also different hotends.. finally tried the firmware, even with multiple tests of Baudrate, but the issues did not go away. 

Finally found the issue that the Print-Overview screen, where you see the progress-bar and temperatures causes micro-stuttering in the printer. When you simply click into one of the menu items on the screen the printing issue disappears. I was using an older version on the firmware from end of 2020, now updated to the latest commit, and the issue is less visible now, but still there.

Here is a timelaps video showing the problem:
At 50% I'm just pressing the "Options" Button and the print issue is gone. When going back to overview, the issue comes back.

https://user-images.githubusercontent.com/1177647/161523951-52d298a7-7254-4c7c-8083-167b8271e0b7.mp4

I'm using the TS35 V2 LCD screen. I'm printing a [simple cube](https://cloud.ikkez.de/index.php/s/dC6d2fY74JypPd7), wall line count: 1, top layers: 0, infill 0%

Print test with latest firmware (commit https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/commit/c51c92124170ccb5da17abbcd694a4a6115aded1) the issue is less but still visible: 

[image]

I could imagine that the change at https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/commit/943658f68449383cf6ad8bf25b24ced00a028b9e in draw_ui.cpp where SPI_QUARTER_SPEED was changed to SPI_FULL_SPEED improved the situation, but I have not validated this.. still issue not gone completely.

Any ideas? Thank you.

MKS-hosiey, 2022-04-07:
Hi, is your gocode sliced with simplify? Have you modified the buffer in the advanced settings?

ikkez, 2022-04-07:
It was sliced with CURA and I doubt it has something to do with the gcode. I have not modified any buffer setting. I did some research I guess that the problem is somewhere in the LVGL implementation.. probably [LVGL-6.1.1-MKS](https://github.com/makerbase-mks/LVGL-6.1.1-MKS) needs to be updated.

ikkez, 2022-04-07:
Another problem that underlines this is that when I touch any button while printing, the printer stops for a moment until the screen is fully rendered.

MKS-hosiey, 2022-04-09:
Can you send your model and firmware to my email? 1428043543@qq.com, I need to test it

MKS-hosiey, 2022-04-11:
Can you provide your email address? I tested with the modified firmware and there is no such issue

ikkez, 2022-04-11:
I have already written you. my email is ikkez0n3@gmail.com ...alternatively try mail@ikkez.de
thanks, I would love to try your version.

MKS-hosiey, 2022-04-11:
> I have already written you. my email is ***@***.*** ...alternatively try ***@***.***

从QQ邮箱发来的超大附件

Mks-Robin-Nano-Marlin2.0-Firmware.zip (243.41M, 2022年05月11日 15:02 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?k=63386439cc5eadc32e0f7a6c4332561c0a59555a52570756145e575f531f500b0c5c49015601521e0a59570956035355010056096517647e524b496b0a500d5d147605570a1f29524b540d57571c541e7f5116541253165617420d49650f&t=exs_ftn_download&code=98d9e2d3

ikkez, 2022-04-12:
no did not work.. i now only see a white screen after updating, like this issue https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/325