makerbase-mks/MKS-Robin-Nano-V3.X #106: Robin nano v3.1 BL Touch failure

Barackazarc, 2022-04-28:
Hello!

I have an issue with this board and antclab bl touch 3.1 (https://www.antclabs.com/bltouch-v3). 

When I try to plug it in and power up the board, the BL Touch literally does nothing. I have checked and the wire that is okay. I have meassured it in multimeter.
Basically when powering on the board the BL Touch does not test itself. Nothing happening.

Any idea how to solve this? I thought it is much easier but it seems I'm having hard time to use this.

Thanks in advance.

MKS-hosiey, 2022-04-30:
The sensor will self-test when it is powered on. It does not self-test. You can contact the seller of the sensor. There should be a problem with the sensor.

mdunfee3016, 2022-07-26:
Verify your wiring connections. This will usually happen if the wires are not connected in the correct order. ALso, what pins are you connecting to?