makerbase-mks/MKS-Robin-Nano-V3.X #109: Disable Sensorless

snow79, 2022-05-07:
Hello Guys
i have MKS robin nano v3.1 and driver TMC2209 in uart mode.
I like to use the standard sensor and disable the sensorless, but i don't find how i can do it.
Please some one can explay how i can do it?
Hardware i have try with end without jumper on diag x, diag y, diag z but nothing change;
on firmware marlin i dont' have enable //#define SENSORLESS_HOMING
the printer don't use sensor and don't stop :(

MKS-hosiey, 2022-05-11:
Is enstop enabled correctly? Is your machine a normal model or a corexy?

AngelBautista, 2022-05-17:
Hello,
I have a MKS robin nano v3.1 and TCM2209 driver, I would like you to help me configure it, it is a home printer, eventually I want to evolve it to x,y. I am a complete novice and I would like to know if by default the sensorless is enabled in the firmware. Thank you very much for your help.

snow79, 2022-05-18:
Step 1 you like to use or not sensorless?If yes you have to enable on firmware and but jumper diag on.If no you have to set on firmware X y limit and remove jumper.For any help I'm hereSent from Android device

AngelBautista, 2022-05-22:
Yes, I would like to use the sensorless option.  which jumper should I use?

[image]

Thank you very much

nfalco79, 2023-12-09:
it's not clear which one is the diag jumper on a Robin nano 3.1