makerbase-mks/MKS-Robin-Nano-V3.X #111: Problems with Marlin with custom FLSUN QQ-S Pro

davedouthettrb, 2022-05-19:
I've spent hours working with FLSUN to fix the firmware for my custom QQ-S Pro with an MKS Robin Nano V3.1 and a TS35 V2.0.

I have the Homing fixed except, if I offset any axis by a small amount manually after homing, Homing fails every time even after a printer reset or power up until I reinstall the firmware.bin file.

Other unresolved issues which either cause build failures or printer function failures:

1.  Auto-leveling crashes into build plate even after homing.
2.  The stock sensor lights up when pressed manually but doesn't get detected by the controller.
3.  My hotend thermistor (from SeeMeCNC) doesn't work correctly (stopped between 160°C and 170°C ... setting 1 & 11 tried).  I'm getting a new one (included with E3D Revo Micro Kit).

A few others have tried the same mods with similar results.

Using the following:
Mks-Robin-Nano-Marlin2.0-Firmware-master.zip
Configurations-release-2.0.9.3.zip (using one of the Delta/FLSUN examples

Someone please help!

Thanks!

MKS-hosiey, 2022-05-23:
Is this the motherboard of the machine? Or did you replace it yourself? This is because your configuration parameters are incorrect. The configuration of delta is very complicated. It is recommended that you look for the configuration file of this machine in the configuration file of marlin.

davedouthettrb, 2022-05-23:
MKS-hosiey,
I replace the original because it failed.A lot of modifications were done to get the Robin Nano and the TS35 in the chassis.
I'm still working on the firmware myself (just finished a build that passed using Configurations-bugfix-2.0.x.zip & Marlin-bugfix-2.0 downloaded today).I've already spent hours on the firmware trying to get everything to work.
I can send you the configuration files to look at if you wish.
Thanks!
Dave

davedouthettrb, 2022-05-23:
MKS-hosiey,

Just tried the new firmware.
1.    Homing is working correctly now.2.    Previously, the touch-screen and decoder were working but now they are erratic.  It takes a bit of effort to get the correct selection and then to activate it.3.    Leveling looked like it was going to work, it stopped moving down about 20mm from the build plate and then move toward the edge of the plate.  Unfortunately, it moved about 5 mm beyond the edge of the build plate and crashed below it.
The sensor lights up on contact with the build plate but the controller doesn't seem to detect it.
If you can help me with this, that would be great!  Files attached.
Thanks!
Dave

 

 
 

    
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

#define CONFIG_EXAMPLES_DIR "delta/FLSUN/QQS-Pro"

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 02010000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some useful links to help get your machine configured and calibrated:
 *
 * Example Configs:     https://github.com/MarlinFirmware/Configurations/branches/all
 *
 * Průša Calculator:    https://blog.prusaprinters.org/calculator_3416/
 *
 * Calibration Guides:  https://reprap.org/wiki/Calibration
 *                      https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 *                      https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 *                      https://youtu.be/wAL9d7FgInk
[... 1400 lines cut ...]
#if ENABLED(PAUSE_BEFORE_DEPLOY_STOW)
  //#define PAUSE_PROBE_DEPLOY_WHEN_TRIGGERED // For Manual Deploy Allenkey Probe
#endif

/**
 * Enable one or more of the following if probing seems unreliable.
 * Heaters and/or fans can be disabled during probing to minimize electrical
 * noise. A delay can also be added to allow noise and vibration to settle.
 * These options are most useful for the BLTouch probe, but may also improve
 * readings with inductive probes and piezo sensors.
 */
//#define PROBING_HEATERS_OFF       // Turn heaters off when probing
#if ENABLED(PROBING_HEATERS_OFF)
  //#define WAIT_FOR_BED_HEATER     // Wait for bed to heat back up between probes (to improve accuracy)
  //#define WAIT_FOR_HOTEND         // Wait for hotend to heat back up between probes (to improve accuracy & prevent cold extrude)
#endif
//#define PROBING_FANS_OFF          // Turn fans off when probing
//#define PROBING_ESTEPPERS_OFF     // Turn all extruder steppers off when probing
//#define PROBING_STEPPERS_OFF      // Turn all steppers off (unless needed to hold position) when probing (including extruders)
//#define DELAY_BEFORE_PROBING 200  // (ms) To prevent vibrations from triggering piezo sensors

// Require minimum nozzle and/or bed temperature for probing
#define PREHEAT_BEFORE_PROBING
#if ENABLED(PREHEAT_BEFORE_PROBING)
  //#d

davedouthettrb, 2022-05-26:
MKS-hosiey,

I keep getting the following error:
arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 176540 bytescollect2.exe: error: ld returned 1 exit status*** [.pio\build\mks_robin_nano_v3_1_usb_flash_drive\firmware.elf] Error 1
What can I do to eliminate this error?
Can I just delete all the disabled features in the Configuration.h file or is there something else I need to change in either the Configuration.h file or the Configuration_Adv.h file?
Thanks!
Dave

 

 
 
 

      
/**   Dave's Mods 25MAY2022 — 09:51PM

 *                      https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 *                      https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 *                      https://youtu.be/wAL9d7FgInk
 *
 *                      https://www.thingiverse.com/thing:1278865
 */

    #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3_1

//#define BAUDRATE_2 115200   // Enable to override BAUDRATE

/**

#define CUSTOM_MACHINE_NAME "FLSUN QQS-Pro MKS Robin Nano 3.1 TS35 V2.0"

 *             https://youtu.be/Bqbcs0CU2FE
 */

#define TEMP_SENSOR_0 5

  #define DELTA_SEGMENTS_PER_SECOND  80

  #define DELTA_PRINTABLE_RADIUS 130.0    // (mm)

  #define DELTA_MAX_RADIUS       130.0    // (mm)

  #define DELTA_HEIGHT 370.00             // (mm) Get this value from G33 auto calibrate

 *   https://reprap.org/forum/read.php?1,739819
 *   https://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html
 */

JoeSiro, 2024-06-14:
hello good morning

were you able to configure the printer?

I have exactly the same electronics as you and I can't get it to work well.

If you can, could you give me the configuration.h and the adv.h, thank you