makerbase-mks/MKS-Robin-Nano-V3.X #116: Robin Nano v3.1 - extruder fan problem

Mauri55, 2022-06-02:
Hello,
I just installed the Robin Nano v3.1 board but i have the following problem:
when the extruder reach the EXTRUDER_AUTO_FAN_TEMPERATURE (set to 50) the extruder fan does not start.

The extruder fan is in connected to FAN1

in pins_MKS_ROBIN_NANO_V3_common.h
#define FAN_PIN                             PC14  // FAN
#define FAN1_PIN                            PB1   // FAN1

in configuration_adv.h
either with //#define E0_AUTO_FAN_PIN -1
or #define E0_AUTO_FAN_PIN -1
the fan does not work

If I send M106 P0 S255 the fan start.

I use the firmware in https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware compiled as per your indication.

Thank you

demirel95, 2022-06-04:
#define E0_AUTO_FAN_PIN PC14 (write which pin it is connected to)

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255  

make settings like this and load like this

sopwith55, 2022-06-07:
Thank you, I'll try as soon as I can !

comtechy14, 2023-07-08:
im also having a problem no matter what I try the extruder fan stays on perm any help would be grateful its robin nano v3

Sherkniem, 2023-09-25:
> Hello, I just installed the Robin Nano v3.1 board but I have the following problem: when the extruder reach the EXTRUDER_AUTO_FAN_TEMPERATURE (set to 50) the extruder fan does not start.
>
> The [...]

Hello! I am experiencing the same issue. I am not able to make it work… Ant help please?

lutherlu, 2023-10-12:
> Hello! I am experiencing the same issue. I am not able to make it work… Ant help please?

Problem solved after I change FAN_PIN to FAN1_pin in pins_MKS_ROBIN_NANO_V3_common.h

mrbrain-true, 2023-12-14:
The right solution to the problem. In file Configuration_adv.h we prescribe the extruder fan:
#define E0_AUTO_FAN_PIN FAN1_PIN