makerbase-mks/MKS-Robin-Nano-V3.X #119: Using 2 fans for blowing

sima35, 2022-06-10:
Well, where can I set up in Martin to run both fans at once and not just one.
 I still have to run the second one manually in marlin.
 I have both fans connected to Fan1 and Fan2.
[image]

sima35, 2022-07-03:
I'm also interested in that, I also have the same problem, for now I have it fixed with one cable, 2 fans

Teius92, 2022-07-20:
I think you have to find the file where there Is the variabile that storie the pwm value of the fan 1 and set the value of pwm fan 2 equal to It. I'm not sure of this but you can try.