makerbase-mks/MKS-Robin-Nano-V3.X #123: Disable boot beep noise

abmantis, 2022-06-30:
Is there any way to disable the boot beep sound (or at least, reduce the volume), without removing the beeper (since I use it for other things)?

abmantis, 2022-07-24:
@makerbase-mks ?

mdunfee3016, 2022-07-26:
[MKS-Robin-Nano-V3.X](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X)/[bootloader](https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/tree/main/bootloader)/No beep/

abmantis, 2022-07-31:
I did that, the printer got bricked and I had to flash using SWD. Now it works, but I can only flash firmware with the USB-A port, not sdcard. 
I guess the bootloader files are not updated for the v3.1 :(

fbeltrao-br, 2023-07-18:
I also want go get rid of this obnoxiously loud beep noise when MCU starts, but @abmantis stated that this bootloader is only for nano 3.0 and bricks 3.1 boards. Is this confirmed do brick nano 3.1 boards?

@makerbase-mks @mks-viva @MKS-Sean @MKS-hosiey please provide an answer.

MKS-hosiey, 2023-07-20:
Both v3.0 and V3.1 are applicable

SnakeKVC, 2024-03-28:
Firmware cannot be installed from micro SD or USB after updating the bootloader to V3.1.

kinsi55, 2024-07-24:
Any update on this? I'd like to disable the boot beep on my board but I wouldnt want to lose the ability of flashing marlin from SD

kinsi55, 2026-02-19:
Its been over a year and this continues to be a problem, I use the no-beep bootloader on my V3.1 but that forces me to always flash via SWD because SD flashes dont work.