makerbase-mks/MKS-Robin-Nano-V3.X #129: Bootloader files bricks V3.1 boards

abmantis, 2022-08-21:
The bootloader files in this repo does not seem to be updated for the v3.1. 
After flashing it, it is not possible to flash firmware using sdcard, only trough the SWD port.

helminec, 2022-09-22:
Unfortunately I made this mistake :(
I have no idea what an SWD port is. Can I save my Nano3.1 and what do I need to do it?
I installed ArduinoIDE.
I can't use any interface on my board, neither USB nor SD card nor UDisk. The printer turns on and the screen says "no connection to printer".

Gixie69, 2023-09-12:
Yeah this has done me to.  It says load the boot loader file and this file and that file and Now i have a  ts35 that flashes black and grey.  These files are from Makerbases own github page. Not being a genius i have no clue how to fix this. More wasted money, more wasted time and a depression level thats about to put me in some serious health troubles.  I came to Makerbase because i liked what they show, Its all so easy.  No no its not. I am furious because yet again.....

SnakeKVC, 2024-01-24:
The bootloader files in this repo does not seem to be updated for the v3.1.
After flashing it, it is not possible to flash firmware using sdcard, only trough the SWD port.

How repair bootloader?