makerbase-mks/MKS-Robin-Nano-V3.X #132: How do I set MKS PWC on the klipper

sima35, 2022-10-15:
Good, I need advice on MKS PWC and klipper.
I have RB Nano v3, the clipper works great, , but I need to start MKS PWC in klipper.
I did not load the information in klipper.
As I understand correctly, MKS PWC is not in their settings.
I would be grateful if you could give me the code for the setup.
After finishing printing, my printer does not turn off automatically.
And I don't know if PWC monitors the voltage.
Or connection to RPI4

sima35, 2022-10-29:
when no one helped us, what could I have thought, I will give you the info here. 
put this in moonraker and watch out for the pin
then proceed as written here in the link> https://www.etechnophiles.com/raspberry-pi-4-gpio-pinout-specifications-and-schematic/#raspberry-pi-4-gpio-pinout

this is how you can control the printer via mainsail

[power Printer]
type: gpio
pin: gpio14
on_when_job_queued: True
off_when_shutdown: True
locked_while_printing: True
restart_klipper_when_powered: false
initial_state: off

ColdBeer72, 2022-12-12:
Hi, @sima35... Is it working for you? I don't find a gpio14 pin in our Robin Nano V3. If you're talking about MKS PWC (not MKS UPS that is a total different think), it need a power detection and a power off connectors: one is for PW_DET pin (PA13 in Klipper) and the other is for PW_OFF pin (PB2 in Klipper).
Is it really working as you told?

ColdBeer72, 2022-12-27:
Ok, friend @sima35 I could make it work without a Raspberry Pi. I explained it on this other issue, **[here](https://github.com/makerbase-mks/Klipper-for-MKS-Boards/issues/18#issuecomment-1365884732)**

sima35, 2023-06-19:
> Ahoj,@sima35... Funguje vám to? V našom Robin Nano V3 nenájdem pin gpio14. Ak hovoríte o MKS PWC (nie o MKS UPS, čo je úplne iný názor), potrebuje detekciu napájania a konektory vypnutia: jeden je pre pin PW_DET (PA13 v Klipper) a druhý je pre pin PW_OFF (PB2 v Klipperi). Naozaj to funguje tak, ako ste povedali?

you need to connect the pw_off cable to the rpi, I have ver.4b