makerbase-mks/MKS-Robin-Nano-V3.X #134: Compilation error - Marlin-2.1.1 - mks_robin_nano_v3_usb_flash_drive - TFT_LVGL_UI

pagabee, 2022-11-06:
It is not possible to compile the actual version of Marlin (2.1.1 aug 7) with the environment mks_robin_nano_v3_usb_flash_drive and  TFT_LVGL_UI.

Marlin\src\lcd\extui\mks_ui\tft_lvgl_configuration.cpp: In function 'void tft_lvgl_init()':
Marlin\src\lcd\extui\mks_ui\tft_lvgl_configuration.cpp:140:77: error: conditional expression between distinct pointer types 'CardReader::sdcard_driver_t*' {aka 'DiskIODriver_SPI_SD*'} and 'DiskIODriver_USBFlash*' lacks a cast
  140 |       card.changeMedia(IS_SD_INSERTED() ? &card.media_driver_sdcard : &card.media_driver_usbFlash);
      |                                                                             ^~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive\src\src\lcd\extui\mks_ui\tft_lvgl_configuration.cpp.o] Error 1

[image]

Valera-Zhan, 2023-02-28:
Hi. 
If the question is still relevant. Required in the file "Marlin\src\lcd\extui\mks_ui\tft_lvgl_configuration.cpp" 
replace line 140: 
```
    #if ENABLED(MULTI_VOLUME) && !HAS_SD_HOST_DRIVE
      SET_INPUT_PULLUP(SD_DETECT_PIN);
      map.changing media(IS_SD_INSERTED() ? and memory cards.media_driver_sdcard : &cards.media_driver_usbFlash);
    #endif
```
with 
```
    #if ENABLED(MULTI_VOLUME) && !HAS_SD_HOST_DRIVE
      SET_INPUT_PULLUP(SD_DETECT_PIN);
      if (read (SD_DETECT_PIN) == Low) in maps.changeMedia(&maps.media_driver_sdcard);
      else card.change the media(&card.media_driver_usb Flash);
    #endif
```

pagabee, 2023-03-02:
Thanks for your code!
I will try it.

menezesfreitas, 2023-05-10:
Também estou com o mesmo problema
[image]

simpit303, 2024-04-02:
Dear MKS.
USB problem is still existing on Marlin bugfixed 2.1.x
You can not compile file as mks_robin_nano_v3_usb_flash_drive_msc with USB_FLASH_DRIVE_SUPPORT defined.
When I turn it off compile is going perfect.