makerbase-mks/MKS-Robin-Nano-V3.X #143: Heated Bed didn't reach the desired temperature - 10ºC difference

jorgejglopez, 2023-01-22:
Hi, I'm having a lot of problems with mi heated bed. I'm using a Robin Nano v3 with a H43 screen and MARLIN-V2.X-MKS-H43-MAIN.

Every time I heat the bed, it didn't reach the temperature. If I set 60ºC, the maximum it reaches is 50ºC. If I set 50ºC, the maximum I reaches is 40ºC. I've disable PID, enable, etc. I've tried with mi Ramps and the heated bed works fine. So I think it's a problem with the firmware but I don't know where it's the problem. I think it's a problem with the configuration because it's always a 10ºC difference between the target temperature and the maximum temperature that the heated bed reach.

ColdBeer72, 2023-01-22:
Maybe you didn't choose the right thermistor type for your hardware setup

jorgejglopez, 2023-01-24:
I've already got it to work correctly. The setup was fine. I was reading in other forums and some people said that resetting the eeprom would solve it. Indeed, by doing a reset to the eeprom and storing the settings, everything has been solved.

closed 2023-01-25 by jorgejglopez