makerbase-mks/MKS-Robin-Nano-V3.X #145: 2004/12864 reprap LCD display not working with robin nano v3

fardeenkhan99, 2023-02-08:
i am trying to use reprap discount smart controller LCD 2004 and 12864 version with MKS robin Nano V3.1 board. i have tested the board using MKS TS35 and the display works fine. the 2004 and 12864 LCD display are also working when used with an arduino and RAMPS 1.6 Board. i have also tried to invert the EXP1 and EXP2 pins, and also placed them upside down. when placed upside down. the display lights up but no text is displayed. i have tried multiple firmware uploads by enabling the displays, but it does not work. am i missing to enable any option in marlin?

boldenko, 2023-04-23:
Same problem occurs, need help

crazyblaze, 2023-05-09:
Hi! I'm with a similar problem. I'm trying to use a LCD 12864 from anet with MKS robin Nano V3 board and it doesn't work it just turns on with a blank screen and the beep seems to work too.... I know the LCD is not broken because it works on a BTT skr 1.4 that I have.

Looking at the board pinout and LCD pinout they seem different to me, but I don't have enough knowledge to further solve this...

[image]
[image]

boldenko, 2023-05-10:
@crazyblaze here's probable solution. MKS lcd pins are rotated to 180 degrees, so if you are using anet LCD you should rotate it and plug. here's the example: https://www.youtube.com/watch?v=k2Ms2Cl5JFQ

crazyblaze, 2023-05-10:
> @crazyblaze here's probable solution. MKS lcd pins are rotated to 180 degrees, so if you are using anet LCD you should rotate it and plug. here's the example: https://www.youtube.com/watch?v=k2Ms2Cl5JFQ
> 
> [image]
> 
> and all the numbers match now!

Hi thank you! But I already did that and it didn't work =`(

wasifshahid, 2023-06-10:
I have found a solution to this problem. 

[image]

Use 

//
// ULTIPANEL as seen on Thingiverse.
//
#define ULTIPANEL

instead of 

//#define REPRAP_DISCOUNT_SMART_CONTROLLER